arpruss/USBComposite_stm32f1 #3: Support for OS to send led status

StarlightNL, 2018-01-02:
I am trying to revive a old keyboard and your library works like a charm but I am missing the ability to get the led status report. Is it implemented and if so how can I get that?

P.S. What is the status on Consumer. Every time I try to send a keypress the stm32(Blue pill) freezes

arpruss, 2018-01-03:
1. Does the LED status report go from the keyboard to the PC or the PC to the keyboard?

2. Did you try the consumer example? It worked for me.

arpruss, 2018-01-03:
Somehow the Consumer.cpp file never got committed to the repository. Fixed.

StarlightNL, 2018-01-03:
From [https://docs.mbed.com/docs/ble-hid/en/latest/api/md_doc_HID.html](https://docs.mbed.com/docs/ble-hid/en/latest/api/md_doc_HID.html) it says that the os is sending an output record to the device. After seeing that I guess that the report descriptors for the keyboard needs to be changed.

arpruss, 2018-01-03:
Added keyboardwithleds example (and updated the library code to allow LEDs to work without constant polling).

closed 2018-01-03 by arpruss