arpruss/USBComposite_stm32f1 #6: USB Keyboard - Multimedia Keys

unic8s, 2018-03-19:
Hi,

I successfully implemented the library and its Keyboard device.
I also want to try to assign the multimedia keys of the v-usb keyboard and found some mappings here https://github.com/arpruss/USBHID_stm32f1/blob/473133ed76e753a02f41b2253134000193f675e8/USBHID.h#L415 but how are they intended to be used from Arduino side? I couldn't figure out how to assign these hex values correctly so that it could be used with the library.

Thanks!

arpruss, 2018-03-19:
You need a Consumer device for these keys. See https://github.com/arpruss/USBHID_stm32f1/blob/master/examples/consumer/consumer.ino

You can have a composite Keyboard and Consumer HID device, but you'll have to construct it. Something like this untested code might work:

    const uint8_t reportDescription[] = {
       HID_KEYBOARD_REPORT_DESCRIPTOR(),
       HID_CONSUMER_REPORT_DESCRIPTOR()
    };

    ...
    USBHID_begin_with_serial(reportDescription, sizeof(reportDescription));

unic8s, 2018-03-19:
Okay, thanks for the quick response. I'll work with that kind of initialization.

Just some moments ago I saw this kind of setup in the 2 joysticks example... :)

closed 2018-03-19 by unic8s