arpruss/USBComposite_stm32f1 #8: Multi serial support

atrips, 2018-05-03:
Hello,

Very nice work !!!

May you please add an example for multiple Serial device support.

Thanks a lot for your effort

arpruss, 2018-05-14:
As in, multiple COM ports?

atrips, 2018-05-14:
Exactly. Like in this project: https://github.com/satoshinm/pill_serial.

arpruss, 2018-05-14:
If you want to add MultipleSerial device support to the library, feel free. It shouldn't be super hard. It's a matter of creating a new plugin, based on the current CompositeSerial plugin.

arpruss, 2018-05-14:
Yes, you're on the right track. 

Make sure you also add the endpoints to serialPartConfigData and add patches to getSerialPartDescriptor() for the new endpoints. The code in usb_generic() modifies the descriptor to allow for compositing with other descriptors, and it does this by patching in various offsets: there are comments in serialPartConfigData as to what is being patched. 

The patching is less important if you're doing serial-only (rather than, say, 2 serial ports + HID), but it'll be more consistent if it's done right. 

You also need to expand the serialEndpoints[] array, and then replace usbSerialPart.endpoints[CDCACM_ENDPOINT_?X] in the tx and rx functions with an index to the correct endpoint. 

You will also want to make vcomData?xCb() functions take an endpoint argument, and add new wrapper callbacks for each endpoint that call the more general function.

arpruss, 2019-12-25:
Multi-serial support (up to 3 ports) has been added.

closed 2019-12-25 by arpruss