arpruss/USBComposite_stm32f1 #9: SDReader Example

Phisatho, 2018-05-16:
The SDReader example does not work for me (on blue pill). It compiles OK, but on one computer, nothing happens. When I plug the board to another laptop, it shows a a drive, when I click the drive, an error pops up.
I verified the connections using an SDFat example.

arpruss, 2018-05-16:
How about the mass example?

You might also want to change the product ID, e.g., CompositeUSB.setProductId(0x0029), before you initialize the mass storage.

Phisatho, 2018-05-17:
You mean USBComposite.setProductId(0x0029). That solved the problem. Is 0x0029 the magic number for USB mass storage? I have tried other numbers before without success.
Mass example works OK. I get an 8kB volatile drive.

Is it possible to make mass storage optional - say based on a button status during reset? I want to make a logger which will be a stand alone device except when connected to a PC for configuration and downloading data.

closed 2018-05-17 by Phisatho

arpruss, 2018-05-17:
I added this to the code.