arpruss/USBComposite_stm32f1 #14: many joysticks in one stm32

vasilyman, 2018-07-26:
Can many joysticks work in one stm32?

vasilyman, 2018-07-26:
Will there be a wireless xbox360 ?

arpruss, 2018-07-28:
I don't know anything about the wireless connections.

The library only supports one xbox controller at a time. It might be possible to extend it to support more than one, but it would require code changes. The easiest, but least elegant, way to add support for one more controller would be to make copies of usb_x360.* and USBXBox360.* and call them usb_x360b.* and USBXBox360b.* and replace all instances of "360" in the two files with "360b". 

The most elegant way would be to modify the generic device implementation to allow multiple instances of the same plugin type.  This would also solve https://github.com/arpruss/USBHID_stm32f1/issues/8

But doing this would increase memory usage for all users, and slightly decrease performance for all users.

arpruss, 2019-12-25:
You could use the trick here: https://github.com/arpruss/GenesisToUSBDual

arpruss, 2019-12-26:
I've just implemented USBMultiXBox360<n>, which allows up to 4 xbox 360 controllers.

I've tried to implement the wireless version, but I can't get Windows to recognize the data from it.

arpruss, 2019-12-28:
Wireless is now implemented.

closed 2019-12-28 by arpruss

vasilyman, 2020-01-02:
Wow! Great news!