arpruss/USBComposite_stm32f1 #15: Porting library to official STM Core

edogaldo, 2018-09-21:
hello @arpruss, did/could you evaluate the porting of your library to the [STM's official core](https://github.com/stm32duino/Arduino_Core_STM32)?

Thanks in advance and best!

arpruss, 2018-09-21:
I would be nice, but no doubt a lot of work. The benefit to me would be that it might then work on the F3.

RickKimball, 2019-01-04:
+1 from me. I just discovered the virtues of this USB implementation. It would be great to have code that worked with both stm32duino and the STM official one.

arpruss, 2019-01-05:
I don't use that core and know very little about it.

rtek1000, 2019-09-25:
https://github.com/stm32duino/Arduino_Core_STM32/issues/668

macbef, 2020-02-28:
Has this awesome lib been ported to STM's stm32duino yet? I tried to compile this but there are still a lot of dependencies to libmaple.
Am I doing something wrong?
I'd like to use roger's core but my bluepill doesn't jump from bootloader (libusb) to HID when I use roger's core. Maybe there's an issue with FS and HS mode for USB but I can't find a solution (in the STM core [using platformio] there is a build flag to set the USB to FS. But not for roger's core.

rtek1000, 2020-03-02:
> 
> I'd like to use roger's core but my bluepill doesn't jump from bootloader (libusb) to HID when I use roger's core.

Maybe you can comment on that in Roger's core issues.

CZHO404, 2020-05-07:
Any progress, I would really like to see this and If no one else does this I might do it my self.

arpruss, 2020-05-07:
You need to port usb_generic.c to do it.

CZHO404, 2020-05-11:
Yea, I've started on a port but Idk if I will ever finish it because I'm still working on other projects. It's actually easier than I originally thought, as ST actually has a USB lib thing called "STM32 USB Device Library" and "STM32 USB Host Library" which are both included with the stm32duino core.

robcazzaro, 2020-09-14:
@CZHO404 any progress by chance? If you don't plan to work on this, can you share what you started doing, so I can see if I might be able to help?

CZHO404, 2020-09-19:
Well I got some success but my computer died so Im kinda back to square 1. I can't even help cause I have nothing to do it with. Sorry when I get everything sorted I might tho.

BennehBoy, 2021-03-12:
bump :D

baniaque, 2021-05-16:
Man, very important library but without official STM Core support is not usable in many projects because most of libs are not compatibile with Roger's core, like LCD and touch screen libs etc.
So please, please, please - make port to official STM Core - much more possibilities and good F4 support :)

czho, 2021-05-17:
> bump :D
> 
> Is everything that needs some port work abstracted into/self contained in usb_generic?
> 
> @czho did you manage to dust off any of the work that you started?

Na my pc completely died saddly