arpruss/USBComposite_stm32f1 #19: boards.h "no such file or directory"

xana43, 2018-12-01:
The boards.h that the entire library is referencing appears to be missing. From what i've checked it seems to be a capitalization error, changing #include boards.h to #include Boards.h in all the header files seems to fix it

arpruss, 2018-12-01:
There must be something wrong with the core installation, because it is "boards.h" in https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/cores/maple

xana43, 2018-12-01:
could be that it's not compatible with my board?

arpruss, 2018-12-01:
What is your board?

closed 2019-01-16 by arpruss

bakoontz2, 2022-07-23:
This appears to have been renamed "board.h" in the most recent STM_32 library found here:

https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json