arpruss/USBComposite_stm32f1 #21: Need help with multiple reports in one Collection(Application)

denisn73, 2019-01-16:
I'm sorry that I write here, just do not know where to write on the subject.

I implemented this descriptor part of FFB and delete all REPORT_ID (have only one ID)
https://github.com/shauleiz/vJoy/blob/master/driver/sys/hidReportDescSingle.h

I initialize data for INPUT report, where "&report" - is joystick control data
HIDReporter(HID, (uint8_t*)&report, sizeof(report), TLID),
It normaly work when manualy sendReport()

How to implement data reception from all other OUTPUT reports and FEATURE reports for so big descriptor with many different reports?

This is my project - [JoyFFB](https://github.com/denisn73/JoyFFB)
This is HID ForceTest programm - [ForceTest.exe](https://github.com/denisn73/JoyFFB/blob/master/ForceTest.exe)

ForceTest normaly work with "vJoy" driver, it see PID and create new effect, but for my joystick have error:
"An invalid parameter was passed to the returning function"
"Error creating <effect name> Effect. Your joystick may not support this effect. "

I assume that it is necessary to write value = 12 in Feature report of "Usage Create New Effect Report"
but i don't understand how make it, and how get data from host

I would be very grateful for the help!

arpruss, 2019-01-16:
I don't think you can have multiple output and feature reports without multiple report ids. I recommend keeping the report ids in place, at least until everything is debugged. It's just one extra byte of memory and data transmission.

I have some examples that involve data going from host to device. Also, I use this functionality to control my Gamecube/Nunchuck to USB adapter: https://github.com/arpruss/gamecube-usb-adapter
and the LEDs on my much simpler SpacePilot to SpaceMouse USB device: https://github.com/arpruss/mouse3d_stm32f1

denisn73, 2019-01-17:
Thank you!!!

Now I managed to get the values of OUTPUT Report.
But I do not understand how to respond to the request FEATURE report ?

USBLyzer show me error on "Control Transfer - Get Report(Feature, Id 23, Len 5)"
This is a report of "Usage PID Pool Report" with FEATURE data.
And i have similar ID in "Usage Set Condition Report" with OUTPUT data.

In programm I add:
> HIDReporter        reporterPool;
> ReportFeature_t   reportPool;
> HIDBuffer_t          dataPoolPID;
> volatile uint8_t bufPoolPID[size]

Initialize the declared variables in class constructor creation:
>reporterPool(*_HID, (uint8_t*)&reportPool, ..., etc),
>dataDeviceGain(bufDeviceGain, ...., etc),

And add buffer to USBHID class
>HID->addFeatureBuffer(&dataPoolPID)

In main loop I check:
> if(reporterPool.getFeature(&value)) {
>            CompositeSerial.println("reporterPool");
>            reporterPool.setFeature((uint8_t*)&bufNewEffect);
>          }

but the condition is not satisfied...
What can be wrong? 

I update my project [JoyFFB](https://github.com/denisn73/JoyFFB)

denisn73, 2019-03-13:
Please help!
I can't process Get Report (FEATURE)
That error from sniffer when host try Get Report:
[image]

ebolzmagy, 2019-04-30:
is possible for FFB device with one endpoint??

arpruss, 2020-01-15:
Sorry to take so long to respond. Don't check for getFeature(): that will only happen when there is a feature OUTPUT (from host). Instead, just do `reporterPool.setFeature((uint8_t*)&bufNewEffect);` whenever bufNewEffect chances, and if the host requests this feature, it should be automatically sent. But I don't know that I've tested feature INPUT.