arpruss/USBComposite_stm32f1 #22: exmples

michalelektryk, 2019-01-16:
So I'm trying to use your library, but only HID examples work. When i upload massStorage sketch my PC says it doesn't recognize the device. Also there are no examples for USB Serial, so I crafted my own code which in my opinion shall work, nut my PC also reports that it didn't recognize the device. I'm using Windows 7 x64 machine and I have plugged ST-link and blue pill at the same time if it makes any difference. Below I also include my code for USB Serial.
In the end I willing to make MIDI and storage device, but for now I'm testing on HID and Serial to exclude some variables.

It would also be nice if you would include some materials about USB in stm32 and in general, so that I wouldn't need to disturb you more often than needed.

[code]
#include <USBComposite.h>

#define LED PC13

USBHID HID;
HIDMouse Mouse(HID);
USBCompositeSerial Ser;

void setup(){
  pinMode(LED,OUTPUT);
  digitalWrite(LED,1);
  HID.begin(HID_MOUSE);
  Ser.begin(115200);
  delay(1000);
}

void loop(){
  digitalWrite(LED,0);
  Mouse.move(4,0);
  delay(500);
  digitalWrite(LED,1);    
  delay(300);
  digitalWrite(LED,0);
  Mouse.move(-4,0);
  delay(500);
  digitalWrite(LED,1);
  delay(300);
  Ser.write("dupa\r\n");
}
[/code]

arpruss, 2019-01-16:
The handy plugin.begin() methods are only designed for use with a single plugin. If you're using both HID and serial, you should do something like:
[code] 
void setup(){
pinMode(LED,OUTPUT);
digitalWrite(LED,1);
HID.setReportDescriptor(HID_MOUSE);
HID.registerComponent();
Ser.registerComponent();
USB.begin();
delay(1000);
}
[/code]

If you want to save a few lines, you can also do this shortcut:
[code] 
void setup(){
pinMode(LED,OUTPUT);
digitalWrite(LED,1);
HID.setReportDescriptor(HID_MOUSE);
HID.begin(Ser, HID_MOUSE);
delay(1000);
}
[/code]

arpruss, 2019-01-16:
It wouldn't surprise me if there were some Win7 problems. The mass storage device is a tiny, tiny ram disk, much smaller than any floppy drive, and it is not surprising if some hosts don't recognize it.

closed 2020-01-15 by arpruss