arpruss/USBComposite_stm32f1 #24: After using the MidiOut.ino example, Board stopped showing as Serial Port.

shaurcasm, 2019-07-28:
I used the MidiOut example and uploaded the program to my STM32F103C8T6 (Black pill) board with roger clark's bootloader to use USB to upload programs. 

The program worked well and with no problems. 

But, the board no longer comes up as a Serial Port to upload programs in the Arduino IDE. It used to be Port 5, but there are none anymore.

1. How can I upload new programs or debug the board now? Is there another way to do so?
2. Else, how can I reset the board to be able to use FTDI to program and keep USB port free for use with this library?

arpruss, 2019-07-28:
Just upload like normal but hit the reset button when the ask is about to upload.

shaurcasm, 2019-07-28:
When I use the USB to upload, it doesn't show up as a port. Only native ports, Port1 and Port2 are shown. 

I even tried to use the the FTDI serial programmer that I used to upload the Roger clark bootloader, but that leads to Failure to Init the device, probably because of the bootloader.

arpruss, 2019-07-28:
The serial port for upload only shows up for a second or two after reset. Have you tried pressing the reset button when the IDE is scanning for the board during the upload sequence?

shaurcasm, 2019-07-28:
The port doesn't show on the tools->Port.

Okay, I tried with native Port 1 and it seemed to connect. But, how do I revert it to original state (Non USB composite) or will I have to constantly upload like this from now on?

shaurcasm, 2019-07-28:
Okay, nevermind my last comment. I uploaded a random sketch and it has reverted back to COM5. 
Thank you for your help! Wasn't able to figure out what to do (Scoured the web before asking you) until you told me about the reset before upload. Good day! 👍

arpruss, 2019-07-28:
Added information on uploading to the readme file.

closed 2019-07-28 by arpruss