arpruss/USBComposite_stm32f1 #25: Unable to change the channel for MidiOut

shaurcasm, 2019-07-29:
Hi! Sorry for another bother. I'm trying to build an e-drum with the STM32F103C8T6. I followed your advice yesterday and I can reprogram the board perfectly now. 

But, a new issue arose for me when I try to change the Midi channel to "10" for the General Midi Drums. I have used this link to set up the rest of the software requirements: https://www.arduino.cc/en/Tutorial/MidiDevice i.e. Using VirtualMIDISynth (with a soundfont FluidR3) and Virtual MIDI Piano Keyboard(VMPK) as the... inter-MIDI-ary software.

I used the example from your library and its working well, playing the notes with the coded velocity. But, I am unable to play drums by changing the channel variable, in the function, to "10". It always plays the same channel, no matter what I change in the VMPK. It's always the Piano.

I used the midiout.ino example given in the USBComposite library provided with the STM32F1 package for Arduino IDE ( From Roger's Git ).

What I have tried thus far:
1. Various numbers in the first variable of the functions like "10", "0x9A", "0x8A" and other arbitrary numbers.
2. Passing the above numbers to the begin function, that setChannel(Unsigned variable)'s the channel. 
3. I even tried to use the MIDIUSB.h just to check if it works after setting the board as a Composite USB. No, it does not. It still doesn't recognise the architecture. 

My layman opinion of what the problem is:
1. I am missing something with software application side of things. Unlikely, but I'm keeping my eyes and ears open for it.
2. I went through the library files to understand them. 
In the file "USBMIDI.h":
In the class USBMIDI, Public section at Line 119: void setChannel(unsigned channel=0);
And further down at Line 135: void begin(unsigned int channel = 0);

Do these functions restrict changing channels by being initialised at 0? If so, how can I change this? When I try to change them locally for eperimenting on the file, I am restricted by a prompt that says "You don't have the permission to open this file. See the owner or Administrator of the file." more or less

TL;DR - Unable to change the channel to Percussion (Channel 10). Need help or advice on how to do the same. Read Last two paragraphs for the core.

fleek, 2019-07-29:
void setChannel(unsigned channel=0);
void begin(unsigned int channel = 0);

These two lines only give a default value of channel 0 if no parameters are given and these two setup only affects the channel in which the device is listening on, for Midi-in purposes.  For your case, the midi.sendNoteOn(0, notes[i], 127);  command affects you, to send a note via channel 10 change the first '0' to '9'

shaurcasm, 2019-07-30:
I am an idiot. Of all the numbers I should have tried. It's working perfectly. Thank you!

closed 2019-07-30 by shaurcasm