arpruss/USBComposite_stm32f1 #28: [Blue Pill STM32F103C8T6] USB Mass Storage W28Q64 (Like a SD card Reader) Failed to format

rtek1000, 2019-09-23:
With the help of another library, it was possible to replace the SD card with a SPI flash memory.

But Windows cannot format the memory.

W25Q64 flash memory is only 8MB (64Mbit) and if the library's own formatter is used, Windows can correctly find the available 8MB space. But when trying to format in Windows 10 x64 Explorer, the size of 4GB appears and then the formatting does not end.

Formatter:
```
// The MIT License (MIT)
// Copyright (c) 2019 Ha Thach for Adafruit Industries

#include "SdFat.h"
#include "Adafruit_SPIFlash.h"

// Use second SPI port
SPIClass SPI_2(2);

#define SS2 PB12

#if defined(__SAMD51__) || defined(NRF52840_XXAA)
Adafruit_FlashTransport_QSPI flashTransport(PIN_QSPI_SCK, PIN_QSPI_CS, PIN_QSPI_IO0, PIN_QSPI_IO1, PIN_QSPI_IO2, PIN_QSPI_IO3);
#else
//  #if (SPI_INTERFACES_COUNT == 1)
//    Adafruit_FlashTransport_SPI flashTransport(SS, &SPI);
//  #else
//    Adafruit_FlashTransport_SPI flashTransport(SS, &SPI);
//  #endif

Adafruit_FlashTransport_SPI flashTransport(SS2, &SPI_2);
#endif

Adafruit_SPIFlash flash(&flashTransport);

// the setup function runs once when you press reset or power the board
void setup()
{
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // wait for native usb

  flash.begin();

  Serial.println("Adafruit Serial Flash Info example");
  Serial.print("JEDEC ID: "); Serial.println(flash.getJEDECID(), HEX);
  Serial.print("Flash size: "); Serial.println(flash.size());
}

void loop()
{
  // nothing to do
}
```
USB Reader:
```
// This uses the greiman sdfat library.
// To use SdFatEX, set ENABLE_EXTENDED_TRANSFER_CLASS to 1 in the library's
// src/SdFatConfig.h
#include <USBComposite.h>
#include <SPI.h>
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"

USBMassStorage MassStorage;
USBCompositeSerial CompositeSerial;

#define LED_PIN PC13
#define PRODUCT_ID 0x29

// Use first SPI port
//SdFat sd;

// Use second SPI port
SPIClass SPI_2(2);
//SdFat sd(&SPI_2);

#define SS2 PB12

Adafruit_FlashTransport_SPI flashTransport(SS2, &SPI_2);

Adafruit_SPIFlash flash(&flashTransport);

//SdFatEX sd;

const uint32_t speed = SPI_CLOCK_DIV2 ;
//const uint8_t SD_CHIP_SELECT = SS2; //PA15; //SS;
bool enabled = false;
uint32 cardSize;

bool write(const uint8_t *writebuff, uint32_t startSector, uint16_t numSectors) {
  //  return sd.card()->writeBlocks(startSector, writebuff, numSectors);
  return flash.writeBlocks(startSector, writebuff, numSectors);
}

bool read(uint8_t *readbuff, uint32_t startSector, uint16_t numSectors) {
  //  return sd.card()->readBlocks(startSector, readbuff, numSectors);
  return flash.readBlocks(startSector, readbuff, numSectors);
}

void setup() {
  //  // Remap SPI 1
  //  // Source: https://community.platformio.org/t/pio-arduino-stm32f103cbt6-remap/6786/5
  //  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY); // PB3 free
  //  afio_remap(AFIO_REMAP_SPI1);
  //
  //  gpio_set_mode (GPIOB, 3, GPIO_AF_OUTPUT_PP);
  //  gpio_set_mode (GPIOB, 4, GPIO_INPUT_FLOATING);
  //  gpio_set_mode (GPIOB, 5, GPIO_AF_OUTPUT_PP);

  USBComposite.setProductId(PRODUCT_ID);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, 1);
}

void initReader() {
  digitalWrite(LED_PIN, 0);
  //  cardSize = sd.card()->cardSize();
  cardSize = flash.size();
  MassStorage.setDriveData(0, cardSize, read, write);
  MassStorage.registerComponent();
  CompositeSerial.registerComponent();
  USBComposite.begin();
  enabled = true;
}

void loop() {
  if (!enabled) {
    //    if (sd.begin(SD_CHIP_SELECT)) {
    if (flash.begin()) {
      initReader();
    }
    else {
      delay(50);
    }
  }
  else {
    MassStorage.loop();
  }
}
```

Adafruit SPI Flash, for FAT filesystems on SPI flash chips:
https://github.com/adafruit/Adafruit_SPIFlash

[Blue Pill STM32F103C8T6] Works!!! USB Mass Storage W28Q64 (Like a SD card Reader):
https://github.com/adafruit/Adafruit_SPIFlash/issues/24#issue-496958474

FirozHemmati, 2023-01-10:
Hello
Thanks for the code written by you
This code works correctly in the first connection and opens the flash drive, but in the second connection and later, that is, after disconnecting and connecting the USB, the flash format mode is restored by Windows.
please guide
  thank you

joesphan, 2024-01-27:
Currently attempting to implement USB mass storage on the STM32 with a spiflash. My understanding is W25QXX chips run at a page size of 256 bytes, meanwhile the standard assumption of sd cards (subsequently sdfat) is a sector size of 512 bytes. This means the read and write would end up with incomplete data.

```cpp
bool write(const uint8_t *writebuff, uint32_t startSector, uint16_t numSectors) {
  //  return sd.card()->writeBlocks(startSector, writebuff, numSectors);
  return flash.writeBlocks(startSector, writebuff, numSectors);
}

bool read(uint8_t *readbuff, uint32_t startSector, uint16_t numSectors) {
  //  return sd.card()->readBlocks(startSector, readbuff, numSectors);
  return flash.readBlocks(startSector, readbuff, numSectors);
}
```

My idea is to have it read 2 pages and write 2 pages (note I've used different libraries, untested code):

```cpp
bool write(const uint8_t *writebuff, uint32_t startSector, uint16_t numSectors) {
  flash.releasePowerDown();
  for (int i = 0; i < numSectors; i++)
  {
    for (int j = 0; j < 256; j++)
    {
      flash.write((startSector + i) * 2, j, *(writebuff + j));
    }
    for (int j = 0; j < 256; j++)
    {
      flash.write(((startSector + i) * 2) + 1, j, *(writebuff + j + 256));
    }

  }
  flash.powerDown();
  return true;
}

bool read(uint8_t *readbuff, uint32_t startSector, uint16_t numSectors) {

  flash.releasePowerDown();
  for (int i = 0; i < numSectors; i++)
  {
    for (int j = 0; j < 256; j++)
    {
      *(readbuff + j) = flash.read((startSector + i) * 2, j);
    }
    for (int j = 0; j < 256; j++)
    {
      *(readbuff + j + 256) = flash.read(((startSector + i) * 2) + 1, j);
    }

  }
  flash.powerDown();
  return true;
}
```

I will report back later.

Regarding the formatting, what is the issue with formatting via windows into a fat16? I don't have a very good understanding of how it all works, but isn't it just reading/writing raw bytes at the end of the day?