arpruss/USBComposite_stm32f1 #33: Multiple USBHID instances ( Multiple HID Devices Mouse and RawHID )

Lan-Hekary, 2019-11-05:
Hello,

I Have an issue with instancing 2 separate HID Devices a Mouse and RawHID.
The code as Follows:
```
#include <USBComposite.h>

#define TXSIZE 256
#define RXSIZE 256

USBHID HID;
HIDRaw<TXSIZE,RXSIZE> raw(HID);

USBHID HIDMouse;
HIDAbsMouse mouse(HIDMouse);

const uint8_t reportDescriptionRaw[] = {
   HID_RAW_REPORT_DESCRIPTOR(TXSIZE,RXSIZE)
};
const uint8_t reportDescriptionMouse[] = {
  HID_ABS_MOUSE_REPORT_DESCRIPTOR(2)
};
void setup(){
  HIDMouse.setReportDescriptor(reportDescriptionMouse, sizeof(reportDescriptionMouse));
  HIDMouse.registerComponent();
  HID.setReportDescriptor(reportDescriptionRaw, sizeof(reportDescriptionRaw));
  HID.registerComponent();
  USBComposite.begin();
  raw.begin();
}

void loop(){
  // Data Gathering and Proccessing then sending via Raw
  //raw.send(data, sizeof(data));
  mouse.move(0,0);
  delay(1000);
  mouse.move(16384,16384);
  delay(1000);
  delay(5);
}
```

I think the problem is that the library sends the last Report Descriptor that was registered via the `setReportDescriptor` method.

There are 2 HID Devices but only the last one is Functional : 
[image]

If I change the order of the 'setReportDescriptor' to report the Mouse last, the HIDRAW device stops and the mouse is instantiated instead like this:
[image]

arpruss, 2019-11-05:
The current code doesn't allow two instances of a single plugin. But you can try to combine the report types and have a composite HID device. However, while that works well for things like joystick and mouse, raw is finicky.

Lan-Hekary, 2019-11-05:
I tried combining the reports but there was issues. With the report ID on Windows ..

I know for a fact that Multiple instances of HID Could be done on Hardware, I've done on ATMEGA32u4 ..
But I am migrating to STM32 for better performance.

Us there anyway to upgrade the library to support this feature ??

arpruss, 2019-11-06:
Not an easy way, because there is currently no way for the same plugin to use two different sets of endpoints. Basically, one would need an almost complete rewrite of the code. I started this once but never finished.

A quick and dirty solution is to copy all the HID plugin code to a new set of files, replacing the word "HID" with "HID2" everywhere (and maybe some other changes will be needed), and then use one HID plugin and one HID2 plugin.