arpruss/USBComposite_stm32f1 #36: Cannot change Joystick HID Descriptor to Add Buttons

robertguyser, 2019-12-11:
I know this is a newbie question, but I am having trouble changing around the USB Joystick descriptor in USBHID.h.

I want to have a joystick with only buttons, and 50 of them. So far I cannot modify the descriptor to have more or less buttons, or anything else, without it crashing the software.

For example, I have tried to decrease the buttons to 16, and modify as such:

```
	0x15, 0x00,						/* 	 Logical Minimum (0) */ \
	0x25, 0x01,						/*    Logical Maximum (1) */ \
	0x75, 0x01,						/*    Report Size (1) */ \
	0x95, 0x10,						/*    Report Count 16) */ \
	0x05, 0x09,						/*    Usage Page (Button) */ \
	0x19, 0x01,						/*    Usage Minimum (Button #1) */ \
	0x29, 0x10,						/*    Usage Maximum (Button #16 ) */ \
	0x81, 0x02,						/*    Input (variable,absolute) */ \
	0x15, 0x00,						/*    Logical Minimum (0) */ \
	0x25, 0x07,						/*    Logical Maximum (7) */ \
	0x35, 0x00,						/*    Physical Minimum (0) */ \
	0x46, 0x3B, 0x01,				/*    Physical Maximum (315) */ \
	0x75, 0x02,						/*    Report Size (2) */ \
	0x95, 0x01,						/*    Report Count (1) */ \
```

But this just results in the code apparently no longer working, as it is not enumerated and my USB serial stops working. 

In short, how does one create a bespoke descriptor?  If this is the wrong place to ask this sort of question, maybe direct me to the proper place? 

Thank you!

arpruss, 2019-12-11:
Did you change the joystick report data structure to match? Are these analog buttons? You give each one a range of values from 0 to 7.

robertguyser, 2019-12-11:
The 0 to 7 was already in the USBHID.H file. I have just started trying to hack at it. What I really want is a joystick with just 50 buttons, no axis or hats or anything.

I am trying to have 50 button outputs from the STM32, they will not be physical buttons in the end, but controlled by a rotary encoder.  I have not changed the report data structure because I am not skilled enough to figure out how. I tried removing just the hat switch, and failed repeatedly.

Where in the data structure is the number of buttons defined? I just see:
```
typedef struct {
    uint8_t reportID;
    uint32_t buttons;
    unsigned hat:4;
    unsigned x:10;
    unsigned y:10;
    unsigned rx:10;
    unsigned ry:10;
    unsigned sliderLeft:10;
    unsigned sliderRight:10;
} __packed JoystickReport_t;
```
and

```
	HIDJoystick(USBHID& HID, uint8_t reportID=HID_JOYSTICK_REPORT_ID) 
            : HIDReporter(HID, (uint8_t*)&joyReport, sizeof(joyReport), reportID) {
        joyReport.buttons = 0;
        joyReport.hat = 15;
        joyReport.x = 512;
        joyReport.y = 512;
        joyReport.rx = 512;
        joyReport.ry = 512;
        joyReport.sliderLeft = 0;
        joyReport.sliderRight = 0;
    }
```

But my attempts to edit this to remove the hat switch failed. Mabye a bit more general advice on an approach to take to achieve this. I was looking at: https://helmpcb.com/electronics/modifying-the-joystick-hid-descriptor and I seem to be editing the descriptor portion in the correct way.

robertguyser, 2019-12-11:
I've tried reducing the logical min/max from those lines to 0,1 - and it still works, but I think that is for the hat switch, yes?

For discussion, here is the complete stock joystick descriptor from USBHID.h as provided:
```
#define HID_JOYSTICK_REPORT_DESCRIPTOR(...) \
	0x05, 0x01,						/*  Usage Page (Generic Desktop) */ \
	0x09, 0x04,						/*  Usage (Joystick) */ \
	0xA1, 0x01,						/*  Collection (Application) */ \
    0x85, MACRO_GET_ARGUMENT_1_WITH_DEFAULT(HID_JOYSTICK_REPORT_ID, ## __VA_ARGS__),  /*    REPORT_ID */ \
	0x15, 0x00,						/* 	 Logical Minimum (0) */ \
	0x25, 0x01,						/*    Logical Maximum (1) */ \
	0x75, 0x01,						/*    Report Size (1) */ \
	0x95, 0x20,						/*    Report Count (32) */ \
	0x05, 0x09,						/*    Usage Page (Button) */ \
	0x19, 0x01,						/*    Usage Minimum (Button #1) */ \
	0x29, 0x20,						/*    Usage Maximum (Button #32) */ \
	0x81, 0x02,						/*    Input (variable,absolute) */ \
	0x15, 0x00,						/*    Logical Minimum (0) */ \
	0x25, 0x07,						/*    Logical Maximum (7) */ \
	0x35, 0x00,						/*    Physical Minimum (0) */ \
	0x46, 0x3B, 0x01,				/*    Physical Maximum (315) */ \
	0x75, 0x04,						/*    Report Size (4) */ \
	0x95, 0x01,						/*    Report Count (1) */ \
	0x65, 0x14,						/*    Unit (20) */ \
    0x05, 0x01,                     /*    Usage Page (Generic Desktop) */ \
	0x09, 0x39,						/*    Usage (Hat switch) */ \
	0x81, 0x42,						/*    Input (variable,absolute,null_state) */ \
    0x05, 0x01,                     /* Usage Page (Generic Desktop) */ \
	0x09, 0x01,						/* Usage (Pointer) */ \
    0xA1, 0x00,                     /* Collection () */ \
	0x15, 0x00,						/*    Logical Minimum (0) */ \
	0x26, 0xFF, 0x03,				/*    Logical Maximum (1023) */ \
	0x75, 0x0A,						/*    Report Size (10) */ \
	0x95, 0x04,						/*    Report Count (4) */ \
	0x09, 0x30,						/*    Usage (X) */ \
	0x09, 0x31,						/*    Usage (Y) */ \
	0x09, 0x33,						/*    Usage (Rx) */ \
	0x09, 0x34,						/*    Usage (Ry) */ \
	0x81, 0x02,						/*    Input (variable,absolute) */ \
    0xC0,                           /*  End Collection */ \
	0x15, 0x00,						/*  Logical Minimum (0) */ \
	0x26, 0xFF, 0x03,				/*  Logical Maximum (1023) */ \
	0x75, 0x0A,						/*  Report Size (10) */ \
	0x95, 0x02,						/*  Report Count (2) */ \
	0x09, 0x36,						/*  Usage (Slider) */ \
	0x09, 0x36,						/*  Usage (Slider) */ \
	0x81, 0x02,						/*  Input (variable,absolute) */ \
    MACRO_ARGUMENT_2_TO_END(__VA_ARGS__)  \
    0xC0
```

Following is what I am pretty sure I am trying to end up with:

```
#define HID_JOYSTICK_REPORT_DESCRIPTOR(...) \
	0x05, 0x01,						/*  Usage Page (Generic Desktop) */ \
	0x09, 0x04,						/*  Usage (Joystick) */ \
	0xA1, 0x01,						/*  Collection (Application) */ \
    0x85, MACRO_GET_ARGUMENT_1_WITH_DEFAULT(HID_JOYSTICK_REPORT_ID, ## __VA_ARGS__),  /*    REPORT_ID */ \
	0x15, 0x00,						/*    Logical Minimum (0) */ \
	0x25, 0x01,						/*    Logical Maximum (1) */ \
	0x75, 0x01,						/*    Report Size (1) */ \
	0x95, 0x32,						/*    Report Count (50) */ \
	0x05, 0x09,						/*    Usage Page (Button) */ \
	0x19, 0x01,						/*    Usage Minimum (Button #1) */ \
	0x29, 0x32,						/*    Usage Maximum (Button #50) */ \
	0x81, 0x02,						/*    Input (variable,absolute) */ \
    MACRO_ARGUMENT_2_TO_END(__VA_ARGS__)  \
    0xC0
```

arpruss, 2019-12-11:
The buttons are in the uint32_t buttons item in the report. That item has room for up to 32 buttons, one per bit. If you want 50 buttons, you will want to add more bits. And you'll probably want to add some padding bits to fill things out to an even number of bytes (56 bits = 7 bytes), or perhaps more easily you could go for 56 buttons and use uint8_t buttons[7].

You might try this: use uint8_t buttons[7] and then just change the 32 report count to 56.

robertguyser, 2019-12-11:
Ok! I will try it now. Thank you!!!

robertguyser, 2019-12-11:
Well. Changing just the descriptor and the struct caused a bunch of errors. Such as:
```
C:\Users\Robert\.platformio\lib\USBComposite for STM32F1/USBHID.h:542:1: error: static assertion failed: Wrong endianness/packing!
 static_assert(sizeof(JoystickReport_t) == 13, "Wrong endianness/packing!");
 ^~~~~~~~~~~~~
C:\Users\Robert\.platformio\lib\USBComposite for STM32F1/USBHID.h: In constructor 'HIDJoystick::HIDJoystick(USBHID&, uint8_t)':
C:\Users\Robert\.platformio\lib\USBComposite for STM32F1/USBHID.h:573:29: error: incompatible types in assignment of 'int' to 'uint8_t [7] {aka unsigned char [7]}'
         joyReport.buttons = 0;
```

I need to change more code so the thing knows how to index the uint8_t buttons[7] data, yes?

arpruss, 2019-12-11:
Right.

robertguyser, 2019-12-11:
Unfortunately that is beyond me at this time. Let's see if I understand what I need to do.

1. (in USBHID.h) Edit the descriptor to have more than 32 buttons reported. This only requires changing Logical Maximum and Usage Maximum. For now I will just switch to 64 buttons so it is a round number of bits.

1. dfsg

arpruss, 2019-12-11:
You can just delete the static assert line of code.

robertguyser, 2019-12-11:
Thank you. That does help. I will add it back in when I figure out how to count bits and know what number it should be - that does a sanity check, yes? I had more typed up and it was lost somehow when I tried to edit it. I am frustrated, but not to the point of mocking your help!!! That was me trying to make the ordered list feature of the message editor to work.

I will go with 64 buttons for the time being, to simplify the discussion. If I cannot reduce them later, that is fine.

joyReport.buttons = 0; now throws an error. I have tried joyReport[i].buttons = 0; in a loop to set all to 0, but this still fails. 

I am sorry this is relatively far beyond my abilities and you have to be the person I harass, but I am fairly determined to move my prototype from an arduino micro to STM32!!

arpruss, 2019-12-11:
What does the new struct look like?

robertguyser, 2019-12-11:
This is what I have now, trying for 64 buttons.

```
typedef struct {
    uint8_t reportID;
    uint8_t buttons[7];
    unsigned hat:4;
    unsigned x:10;
    unsigned y:10;
    unsigned rx:10;
    unsigned ry:10;
    unsigned sliderLeft:10;
    unsigned sliderRight:10;
} __packed JoystickReport_t;

//static_assert(sizeof(JoystickReport_t)==13, "Wrong endianness/packing!");

class HIDJoystick : public HIDReporter {
protected:
	JoystickReport_t joyReport; 
    bool manualReport = false;
	void safeSendReport(void);
public:
	inline void send(void) {
        sendReport();
    }
    void setManualReportMode(bool manualReport); // in manual report mode, reports only sent when send() is called
    bool getManualReportMode();
	void begin(void);
	void end(void);
	void button(uint8_t *button, bool val);
	void X(uint16_t val);
	void Y(uint16_t val);
	void position(uint16_t x, uint16_t y);
	void Xrotate(uint16_t val);
	void Yrotate(uint16_t val);
	void sliderLeft(uint16_t val);
	void sliderRight(uint16_t val);
	void slider(uint16_t val);
	void hat(int16_t dir);
	HIDJoystick(USBHID& HID, uint8_t reportID=HID_JOYSTICK_REPORT_ID) 
            : HIDReporter(HID, (uint8_t*)&joyReport, sizeof(joyReport), reportID) {
        joyReport.buttons[0] = 0;
        joyReport.buttons[1] = 0;
        joyReport.buttons[2] = 0;
        joyReport.buttons[3] = 0;
        joyReport.buttons[4] = 0;
        joyReport.buttons[5] = 0;
        joyReport.buttons[6] = 0;
        joyReport.buttons[7] = 0;      
        joyReport.hat = 15;
        joyReport.x = 512;
        joyReport.y = 512;
        joyReport.rx = 512;
        joyReport.ry = 512;
        joyReport.sliderLeft = 0;
        joyReport.sliderRight = 0;
    }
};
```

Then in joystick.cpp:
```
void HIDJoystick::button(uint8_t *button, bool val){
	uint32_t mask = ((uint32_t)1 << (button-1));

	if (val) {
        joyReport.buttons |= mask;
	} else {
        joyReport.buttons &= ~mask;
[... 40 lines cut ...]
     Joystick.button(i,(0,0,0,0,0,0,0,0));
                              ^
src\main.cpp:37:32: warning: right operand of comma operator has no effect [-Wunused-value]
     Joystick.button(i,(0,0,0,0,0,0,0,0));
                                ^
src\main.cpp:37:34: warning: right operand of comma operator has no effect [-Wunused-value]
     Joystick.button(i,(0,0,0,0,0,0,0,0));
                                  ^
src\main.cpp:37:36: warning: right operand of comma operator has no effect [-Wunused-value]
     Joystick.button(i,(0,0,0,0,0,0,0,0));
                                    ^
src\main.cpp:37:38: warning: right operand of comma operator has no effect [-Wunused-value]
     Joystick.button(i,(0,0,0,0,0,0,0,0));
                                      ^
src\main.cpp:37:40: error: invalid conversion from 'int' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
     Joystick.button(i,(0,0,0,0,0,0,0,0));
                                        ^
In file included from C:\Users\Robert\.platformio\lib\USBComposite for STM32F1/USBComposite.h:10:0,
                 from src\main.cpp:2:
C:\Users\Robert\.platformio\lib\USBComposite for STM32F1/USBHID.h:539:7: note:   initializing argument 1 of 'void HIDJoystick::button(uint8_t*, bool)'
  void button(uint8_t *button, bool val);
       ^~~~~~
*** [.pio\build\genericSTM32F103C8\libd53\USBComposite for STM32F1\Joystick.cpp.o] Error 1
*** [.pio\build\genericSTM32F103C8\src\main.cpp.o] Error 1
```

robertguyser, 2019-12-11:
Got it working! At least with 64 buttons.. Now I can try to make a new hid descriptor altogether for just my device and add it in. thank you for your help. I am sure my solution is crude and hackish, and any advice to improve it is welcome!

Here is what worked to get 64 buttons:

USBHID.h
```
//Descriptor snippet
	0x75, 0x01,						/*    Report Size (1) */ \
	0x95, 0x40,						/*    Report Count (64) */ \
	0x05, 0x09,						/*    Usage Page (Button) */ \
	0x19, 0x01,						/*    Usage Minimum (Button #1) */ \
	0x29, 0x40,						/*    Usage Maximum (Button #64) */ \

// Report struct
typedef struct {
    uint8_t reportID;
    uint8_t buttons[8];
    unsigned hat:4;
    unsigned x:10;
    unsigned y:10;
    unsigned rx:10;
    unsigned ry:10;
    unsigned sliderLeft:10;
    unsigned sliderRight:10;
} __packed JoystickReport_t;

//static_assert(sizeof(JoystickReport_t)==13, "Wrong endianness/packing!");

class HIDJoystick : public HIDReporter {
protected:
	JoystickReport_t joyReport; 
    bool manualReport = false;
	void safeSendReport(void);
public:
	inline void send(void) {
        sendReport();
    }
    void setManualReportMode(bool manualReport); // in manual report mode, reports only sent when send() is called
    bool getManualReportMode();
	void begin(void);
	void end(void);
	void button(uint8_t button, bool val);
	void X(uint16_t val);
	void Y(uint16_t val);
	void position(uint16_t x, uint16_t y);
	void Xrotate(uint16_t val);
	void Yrotate(uint16_t val);
	void sliderLeft(uint16_t val);
	void sliderRight(uint16_t val);
	void slider(uint16_t val);
	void hat(int16_t dir);
	HIDJoystick(USBHID& HID, uint8_t reportID=HID_JOYSTICK_REPORT_ID) 
            : HIDReporter(HID, (uint8_t*)&joyReport, sizeof(joyReport), reportID) {
        joyReport.buttons[0] = 0;
        joyReport.buttons[1] = 0;
        joyReport.buttons[2] = 0;
        joyReport.buttons[3] = 0;
        joyReport.buttons[4] = 0;
        joyReport.buttons[5] = 0;
        joyReport.buttons[6] = 0;
        joyReport.buttons[7] = 0;      
        joyReport.hat = 15;
        joyReport.x = 512;
        joyReport.y = 512;
        joyReport.rx = 512;
        joyReport.ry = 512;
        joyReport.sliderLeft = 0;
        joyReport.sliderRight = 0;
    }
};
```

joystick.cpp
```
void HIDJoystick::button(uint8_t button, bool val){
	
    uint8_t mask;
    uint8_t val2 = 0;

    for(int i=0; i <= 7; i++){
        val2 = i;
        mask = ((uint8_t)1 << (button-(1+(i*8))));
	if (val) {
        joyReport.buttons[val2] |= mask;
	} else {
        joyReport.buttons[val2] &= ~mask;
	}
    }    
	
    safeSendReport();
}
```

robertguyser, 2019-12-11:
I am going to close the issue and conclude with a feature request:

It would be great if there was an easier way to pass configuration to the descriptor from the main program, and attendant refactoring of joystick.cpp to work with arrays of values instead of just values.  For example, the popular arduino HID joystick library has the following constructor:
```
Joystick_ SimKnob1(JOYSTICK_DEFAULT_REPORT_ID,
                   JOYSTICK_TYPE_GAMEPAD, 50, 0,
                   false, false, false, false, false, false,
                   false, false, false, false, false);
```

 I might attempt it, but it is likely outside my abilities to do quickly.

closed 2019-12-11 by robertguyser

arpruss, 2019-12-11:
That would take a lot of refactoring, I think, in that the Joystick, and also presumably Keyboard and Mouse as well, class would have to be helping to dynamically construct the HID report descriptor. Currently, the HID report descriptor is statically constructed, which reduces memory usage and simplifies the code.

One could write a wrapper library that does this. That would be nice.