arpruss/USBComposite_stm32f1 #44: This should check for 8bytes alignment as per reference manual.

victorpv, 2020-01-08:
https://github.com/arpruss/USBComposite_stm32f1/blob/90ab85438e5ddc98ab04e13ab1859b80f9ad7c8d/USBCompositeSerial.h#L40

This is regarding the discussion we had over the libmaple pma<>buffer copy functions. As per STM reference manual buffer size ( and buffer start position) should have 8 byte alignments. Just wanted to make the note here in the part that sets the buffer size.

arpruss, 2020-01-08:
Are we talking about alignment of PMA buffers here?

victorpv, 2020-01-08:
Yes

arpruss, 2020-01-08:
It's a bit more complicated by my reading of the reference manual: if the size of the buffer is less than 64 bytes, the length must be even; if the size of the buffer is 64 bytes or more, it must be divisible by 32.  I fixed this in the abstracted branch, which soon will supercede master.

victorpv, 2020-01-09:
Even more complicated from what I see, the 2 / 32 multiplier thing only applies to the RX buffers, but not the TX...
Glad that you figured it out.

arpruss, 2020-01-09:
Yeah. I'll revert for tx.
I'm now dealing with an odd issue where serial fails unless the buffer size is a power of 2.

closed 2020-01-09 by arpruss