arpruss/USBComposite_stm32f1 #47: RawHid example

nikthefix, 2020-01-15:
Hi, Using Simple HID Write utility and other similar utilities I find that the looped back data are incremented by 1 - so if I send 01, 02, 03, 04 hex the maplemini sends back 02, 03, 03, 04..
I've looked through the libs and I can't see the source of this shift. Could you point me in the right direction? If I send buffered data one-way from maple to pc then all is fine so my issue seems to be maple reception of SET REPORT. 

Thanks for the great work.

arpruss, 2020-01-15:
Are you sending a packet of exactly 300 bytes?

nikthefix, 2020-01-16:
I'm limited to 64 bytes for rx and tx buffers (Simple HID Write creates exception if buffers are set higher) and sending 64+1 reports. I've tried 4 byte reports too. I'm using Windows 10. Any unused report bytes are stuffed with '01' and not '00'.

Thanks

nikthefix, 2020-01-16:
Additionally, all your other examples using any form of HID work perfectly.

nikthefix, 2020-01-17:
Solved:

The problem was line 55 in the sketch

for (int i=0;i<RXSIZE;i++) buf[i]++;

It's not needed.

arpruss, 2020-01-17:
Yes, this is intentional: it's to check that the host side isn't just listening to itself, but is actually receiving processed data from the other side.

closed 2020-01-17 by arpruss

nikthefix, 2020-01-17:
Ah I see. Terrific. Thanks.