arpruss/USBComposite_stm32f1 #48: MIDI + Keyboard + Serial

PavelLazarevd, 2020-01-15:
How can i do this 

	HID.begin(HID_KEYBOARD);
	
	midi.registerComponent();
	CompositeSerial.registerComponent();

?

arpruss, 2020-01-15:
```
HID.setReportDescriptor(HID_KEYBOARD);
HID.registerComponent();
CompositeSerial.registerComponent();
USBComposite.begin();
```

The `HID.begin()` shortcut should only be used for a single-component device.

But note that at least on Win10, a composite device with MIDI and serial in it won't have the serial work.

closed 2020-01-15 by arpruss

PavelLazarevd, 2020-01-15:
Ок and other question 
How can i create custom  i have test code

int main() {
	struct usb_bus *busses;
	usb_init();
	usb_find_busses();
	usb_find_devices();
	busses = usb_get_busses();
	struct usb_bus *bus;
	struct usb_device * dev = NULL;
	for (bus = busses; bus; bus = bus->next) {
		for(dev = bus->devices; dev; dev = dev->next) {
			if(dev->descriptor.idVendor == 0xCAFE && dev->descriptor.idProduct == 0xC002) {
				//Нашли первое устройство. Работаем с ним. Если нужно работать со всеми, то собираем список здесь
				goto found;
			}
		}
	}

	fprintf(stderr, "Not found devices\n");
	fflush(stderr);
	return __LINE__;

found:
	;

	usb_dev_handle *ukey_handle =  usb_open(dev);
	if (NULL == ukey_handle) {
	  fprintf(stderr, "Error opening: %s\n", usb_strerror());
		fflush(stderr);
		return __LINE__;
	}

	const int endpoint_out_num = 3; //endpoint для посылки команд
	const int endpoint_in_num = 2;  //endpoint для чтения ответов
	//По хорошему нужно искать нужные номера endpoint в interface discriptor, но т.к. устройство у нас константное, искать не будем

	if(1){
		char beeps_list[] = "0123456789";

		for(unsigned int i=0; i < strlen(beeps_list); i++){
			char cmd = beeps_list[i];
			int tryes;
			for(tryes = 5; tryes; tryes--){
				int slen = usb_bulk_write(ukey_handle, endpoint_out_num, &cmd, sizeof(cmd), 5000);

				printf("sended %i ", slen);

				if (slen <= 0) {
					printf("\n");
					usleep(1000000UL);// подождём немного. У клавиатуры буфер приёма всего один или два пакета. Предыдущий ещё не обработался.
					continue;
				}

				print_dump((unsigned char*)&cmd, 0, sizeof(cmd));
				printf("\n");
				break;
			}

			if(!tryes){
				fprintf(stderr, "Not send command\n");
				fflush(stderr);
			}
		}
	}

	if(1){ //request version
		int tryes;
		for(tryes = 5; tryes; tryes--){
			char cmd[] = "\x3f";
			int slen = usb_bulk_write(ukey_handle, endpoint_out_num, cmd, strlen(cmd), 5000);

			printf("sended %i\n", slen);

			if (slen <= 0) {
				usleep(1000000UL);// подождём немного. У клавиатуры буфер приёма всего один или два пакета. Предыдущий ещё не обработался.
				continue;
			}

			if(1){// Нет смысла читать ответ здесь, т.к ответ приходит ещё и в HID интерфейс
			  char resp[8];
			  int rlen = usb_bulk_read(ukey_handle, endpoint_in_num, resp, sizeof(resp), 1000);

				printf("recvd %i ", rlen);

				if (rlen <= 0) {
					printf("\n");
					continue;
				}

				if (rlen > int(sizeof(resp))) {
					rlen = sizeof(resp);
				}

				print_dump((unsigned char*)&resp, 0, rlen);
				printf("\n");
				break;

			} else {
				break; // Не читаем. Ответ придёт в HID
			}
		}

		if(!tryes){
			fprintf(stderr, "Not send command\n");
			fflush(stderr);
		}
	}

	return 0;
}

need send some command on STM32

arpruss, 2020-01-15:
I don't understand what you are trying to do. Sorry.

PavelLazarevd, 2020-01-15:
need create device  HID USB KEYBOARD with custom usb functions (when i send usb_bulk_write on pc stm doing my code (playing notify sound))
i can`t change windows driver they 
need create custom usb with 2 endpoint 
const int endpoint_out_num = 3; //endpoint 
const int endpoint_in_num = 2;  //endpoint 
and process input command and out command

[lsusb.txt](https://github.com/arpruss/USBComposite_stm32f1/files/4065872/lsusb.txt)
[device-descriptors.hex.txt](https://github.com/arpruss/USBComposite_stm32f1/files/4065874/device-descriptors.hex.txt)
[hidreport-descriptor-hex.txt](https://github.com/arpruss/USBComposite_stm32f1/files/4065875/hidreport-descriptor-hex.txt)

arpruss, 2020-01-15:
You need to make your own plugin. Start by taking the plugin closest to what you want. 

The way I write plugins is in two parts: there is a low level part in C (e.g., usb_composite_serial.c) and a high level part in C++ (e.g., USBCompositeSerial.cpp). 

There is no documentation on how to do it.

PavelLazarevd, 2020-01-15:
Саn you help me with plugin? i dont know how i`ll make it self

arpruss, 2020-01-15:
Sorry: it would take too much hand-holding. I suggest you modify an old one.

PavelLazarevd, 2020-01-15:
I don't have an old source, I'm remaking the device.

arpruss, 2020-01-15:
I meant: take the source code for one of the plugins in this library and change it to fit your needs.

PavelLazarevd, 2020-01-15:
Well I'll try