arpruss/USBComposite_stm32f1 #51: Serial port & Mass storage

rapzak, 2020-01-25:
Hi,

What do i do wrong in this code.
I am trying to have a  serial port and a Mass storage.
If i run mass or serial sepperate it seems to work.
Running this code her without toching pc, it will disconnect after short time, and re-enumerate again, but wrong.
Mass seems to work until re-enumeration, but serial will not connect.

Thank you.

`#include <Arduino.h>
#include <USBComposite.h>
#include <SPI.h>
#include "SdFat.h"

#define PRODUCT_ID 0x30

USBMassStorage MassStorage;
USBCompositeSerial CompositeSerial;
SdFat sd; //ex
const uint32_t speed = SPI_CLOCK_DIV2 ;
const uint8_t SD_CHIP_SELECT = SS;
bool enabled = false;
uint32 cardSize;

bool write(const uint8_t *writebuff, uint32_t startSector, uint16_t numSectors) {
  return sd.card()->writeBlocks(startSector, writebuff, numSectors);
}

bool read(uint8_t *readbuff, uint32_t startSector, uint16_t numSectors) {
  return sd.card()->readBlocks(startSector, readbuff, numSectors);
}

void setup() {
  USBComposite.setProductId(PRODUCT_ID);
}

void initReader() {
  cardSize = sd.card()->cardSize();  
  MassStorage.setDriveData(0, cardSize, read, write);
  USBComposite.clear();
  MassStorage.registerComponent(); 
  CompositeSerial.registerComponent();
  USBComposite.begin();
  while (!USBComposite);
  enabled=true;
}

void loop() {
  uint32_t c = 0;
  if (!enabled) {
    if (sd.begin(SD_CHIP_SELECT)) {
      initReader();
    }
    else {
      delay(50);
    }
  }
  else {
    c++;
    MassStorage.loop(); 
    if (c>10000){
      if(CompositeSerial.isConnected())
        CompositeSerial.println("test"); 
      c = 0;
      }
  }
}`

arpruss, 2020-01-25:
Have you tried swapping the order of the two component registrations?

rapzak, 2020-01-25:
Hi,
Thanks for answer.

Yes, that seems equal.

After re-enumerate, device manager tels:
Unknown usb device (Device descriptor request failed)

/Kasper

```
void initReader() {
  cardSize = sd.card()->cardSize();  
  MassStorage.setDriveData(0, cardSize, read, write);
  USBComposite.clear();
  CompositeSerial.registerComponent();
  MassStorage.registerComponent(); 
  USBComposite.begin();
  while (!USBComposite);
  enabled=true;
}
```

arpruss, 2020-01-25:
Can you try the 0.93 version?

How long before the re-enumerate?

rapzak, 2020-01-25:
0.93 seems to behave more or less equal also.

45seconds if i do not open file/serial

had to change the wait for composite is ready...
```
void initReader() {
  cardSize = sd.card()->cardSize();  
  MassStorage.setDriveData(0, cardSize, read, write);
  USBComposite.clear();
  CompositeSerial.registerComponent();
  MassStorage.registerComponent(); 
  USBComposite.begin();
  //while (!USBComposite);
  delay(50);
  enabled=true;
}
```

rapzak, 2020-01-25:
0.9 seems also to behave more or less equal also.

rapzak, 2020-01-25:
Running on this board:
http://www.keil.com/mcbstm32/mcbstm32-schematics.pdf
the Jumper at VUSB(PD2) is set to GND.

rapzak, 2020-01-26:
Updated the core to latest from master - problem is still the same.