arpruss/USBComposite_stm32f1 #52: USB_EP_TYPE_ISO

rapzak, 2020-01-25:
I have to make this to usb_audio.h::
#define USB_EP_TYPE_ISO USB_EP_T_ISO

I am not sure why...

/Kasper

C:\Users\xxxx\.platformio\lib\USBComposite for STM32F1_ID2052\usb_audio.c:227:32: error: 'USB_EP_TYPE_ISO' undeclared here (not in a function); did you mean 'USB_EP_T_ISO'?
         .bmAttributes        = USB_EP_TYPE_ISO, /* isochronous, not shared */
                                ^~~~~~~~~~~~~~~
                                USB_EP_T_ISO

arpruss, 2020-01-25:
It seems to be define here in the core:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/2ecec5268ba60ab04422448789a9975c3d31ca8f/STM32F1/system/libmaple/include/libmaple/usb.h

rapzak, 2020-01-26:
Correct - after i updated the core..

closed 2020-01-26 by rapzak