arpruss/USBComposite_stm32f1 #54: getMIDIPartDescriptor setting wrong packet size

dewhisna, 2020-02-19:
The `getMIDIPartDescriptor` function in `usb_midi_device.c` is setting both the Rx and Tx endpoints of the descriptor to the same `usb_midi_txEPSize` value instead of setting the Rx to `rxEPSize`:
https://github.com/arpruss/USBComposite_stm32f1/blob/e9a94634505abf4cb4b7c3577527c4465627d1d6/usb_midi_device.c#L288-L289

<br/>This contradicts the logic of `usb_midi_setTXEPSize` and `usb_midi_setRXEPSize` and having separate Rx and Tx sizes:
https://github.com/arpruss/USBComposite_stm32f1/blob/e9a94634505abf4cb4b7c3577527c4465627d1d6/usb_midi_device.c#L304-L318

<br/>It seems line 289 should change as:
```diff
---    OUT_16(usbMIDIDescriptor_Config, DataOutEndpoint.wMaxPacketSize) = usb_midi_txEPSize;
+++    OUT_16(usbMIDIDescriptor_Config, DataOutEndpoint.wMaxPacketSize) = rxEPSize;
```

<br/>I don't see any reason it would be using the same Tx size for both and seems to be working only because the Rx and Tx sizes are generally set to the same value.  Though on the other hand, if there is a reason for them to be the same that I'm not seeing, it should at least be appropriately commented in the code.

arpruss, 2020-02-19:
Fixed. Thanks!

closed 2020-02-19 by arpruss

dewhisna, 2020-02-19:
Thanks, but does your change compile?  For some reason, there appears to be an inconsistency in the naming of the Rx and Tx buffer size variables.  The Tx is called `usb_midi_txEPSize` (and is global instead of static) and the Rx is called `rxEPSize` and is static to this code file.  So I would have thought you'd need `rxEPSize` there instead of `usb_midi_rxEPSize`.  Or better yet, fix the naming of the Rx side so that it's `usb_midi_rxEPSize` everywhere in this file. ;-)

arpruss, 2020-02-19:
Good catch. Fixed again. :-)

I think what happened was that the MIDI code I started with just had rxEPSize and txEPSize. Which isn't what you want in the global namespace. But only the tx needs to be accessible from outside.

dewhisna, 2020-02-19:
Yep, I saw that the Tx was used in other code files (hence why not static) and I figured that was the case (without digging into the history).  But it is still probably best to have a consistent naming convention, even if the Rx remains static.

I happened to stumble on all of this because I'm reworking this part on my fork so I can have multiple MIDI ports and additional IN/OUT jacks, etc., as I'm building a fairly large MIDI Interface box.  The first iteration will have 3 complete MIDI interfaces (In/Out/Thru on each) and a future version (that bumps from a STM32F103CB to either a STM32F103VG or STM32F405VG) will have 6 ports.  And since you can't have multiple instances of this device, I'm doing quite an extensive code rework on the MIDI portion and stumbled on this in the process.