arpruss/USBComposite_stm32f1 #55: Compile error on USB_EP_TYPE_ISO in usb_audio.c

3gyptian, 2020-02-28:
Downloaded the latest but not compiling ...

```Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.1.4
|-- <USBComposite for STM32F1> 0.95
Building in release mode
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_audio.c.o
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_generic.c.o
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_hid.c.o
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_mass.c.o
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_mass_mal.c.o
lib/USBComposite_stm32f1/usb_audio.c:225:32: error: 'USB_EP_TYPE_ISO' undeclared here (not in a function); did you mean 'USB_EP_T_ISO'?
         .bmAttributes        = USB_EP_TYPE_ISO, /* isochronous, not shared */
                                ^~~~~~~~~~~~~~~
                                USB_EP_T_ISO
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_midi_device.c.o
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_multi_serial.c.o
*** [.pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_audio.c.o] Error 1
Compiling .pio/build/genericSTM32F103CB/lib6a2/USBComposite_stm32f1/usb_multi_x360.c.o```

I'm running platformIO and it compiles OK if I revert back to the maple core version which is working fine. I only upgraded to pick up the midi fix I noticed was just fixed last week.

Awesome library .. thank you for this!

3gyptian, 2020-03-06:
Was able to compile by modifying instances of USB_EP_TYPE_ISO to be USB_EP_T_ISO in usb_audio.c

arpruss, 2020-03-07:
What core are you compiling against?

USB_EP_T_ISO is not the same value as USB_EP_TYPE_ISO.

arpruss, 2020-03-08:
USB_EP_TYPE_ISO is defined in [usb.h](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/2ecec5268ba60ab04422448789a9975c3d31ca8f/STM32F1/system/libmaple/include/libmaple/usb.h), which should be included by usb_audio.c. Can you check if usb_audio.c is correctly including usb.h?

3gyptian, 2020-03-09:
That was it .. thank you!!  It turns out PlatformIo is no longer updating the maple core and it's stuck at v1.10 .. an old version from about a year ago and the usb.h did not contain USB_EP_TYPE_ISO.   After upgrading the core the compile completes without error.

It was surprisingly painful to upgrade the maple core to the latest as if you simply git clone Roger's repo, the PlatformIO code detects missing package.json and downloads v1.10 again stomping on the manual upgrade.

For anyone who's interested here were the steps I took to manually upgrade the maple core:

```
# cd ~/.platformio/packages/
# mv framework-arduinoststm32-maple junk
# git clone https://github.com/rogerclarkmelbourne/Arduino_STM32.git
# mv Arduino_STM32 framework-arduinoststm32-maple
# cd framework-arduinoststm32-maple
# cp ../junk/package.json .    
# cd tools
# cp ../../junk/tools/plat*py .
# rm -rf ../../junk

```

Some help came from here:  https://community.platformio.org/t/stm32-maple-update/11690

DeepSouthMan, 2020-08-07:
> That was it .. thank you!! It turns out PlatformIo is no longer updating the maple core and it's stuck at v1.10 .. an old version from about a year ago and the usb.h did not contain USB_EP_TYPE_ISO. [...]

This worked like a champ! I was working on this error for the last 2 days!
I really appreciate your hard work.

CodeOhms, 2020-11-02:
I tried doing this work around but no matter what, Platformio always finds a way to reinstall the old files :angry:!
I even tried setting the folder to read only, so it just deleted the folder and reinstalled it!
I will try one more thing which is to set the owner as root...

CodeOhms, 2020-11-02:
I figured out my own workaround, which involves adding a new framework by editing some Platformio files. I followed this official guide to help me defeat this :trollface: of a problem [:link:](https://docs.platformio.org/en/latest/platforms/creating_platform.html).
**I RECOMMEND YOU BACKUP ALL FILES YOU EDIT!**

## First of all, enter the Platformio files directory and get the framework:
```console
cd ~/.platformio/packages/
git clone https://github.com/rogerclarkmelbourne/Arduino_STM32.git framework-arduinoststm32-mapleUPDATED
cd framework-arduinoststm32-mapleUPDATED
git submodule init
git submodule update
```

## Now we need to edit some files:
    Add this to your project's `Platformio.ini` file:
```ini
board_build.core = mapleUPDATED
```

It should look something like this:
```ini
[env:bluepill_f103c8_128k]
platform = ststm32
board = bluepill_f103c8_128k
framework = arduino
board_build.core = mapleUPDATED
lib_deps = arpruss/USBComposite for STM32F1@^0.99
```

Now we need to edit `platform.json`
```console
cd ~/.platformio/platforms/ststm32
```

Add these lines under `"packages": { ...`
```json
"framework-arduinoststm32-mapleUPDATED": {
        "type": "framework",
        "optional": true,
        "version": "~2.10000.0"
},
```

Edit `platform.py`:
Search for `if "arduino" in frameworks:` at the top of the file. Copy the following into the if statement for `if build_core == "maple"`:
```python
self.packages["framework-arduinoststm32-mapleUPDATED"]["optional"] = True
```

and beneath that if statement block copy the following:
```python
if build_core == "mapleUPDATED":
                self.frameworks["arduino"]["package"] = "framework-arduinoststm32-mapleUPDATED"
                self.packages["framework-arduinoststm32-mapleUPDATED"]["optional"] = False
                self.packages["framework-arduinoststm32-maple"]["optional"] = True
                self.packages["framework-arduinoststm32"]["optional"] = True
```

Now, from `~/.platformio/platforms/ststm32` cd into `builder/frameworks`.
Edit `arduino.py`, adding this below the if statment block `if core == "maple:":
```python
if core == "mapleUPDATED":
    build_script = join(
        env.PioPlatform().get_package_dir("framework-arduinoststm32-mapleUPDATED"),
        "tools", "platformio-build-%s.py" % mcu[0:7])
```

Next we need to edit some of the files for our updated core package:
```console
cd ~/.platformio/packages/framework-arduinoststm32-mapleUPDATED/
```

and add a file named `package.json`:
```json
{
  "name": "framework-arduinoststm32-mapleUPDATED",
  "version": "2.10000.200103",
  "description": "Arduino Wiring-based Framework for ST STM32 microcontrollers (Maple Core)",
  "keywords": [
    "framework",
    "arduino",
    "stmicroelectronics",
    "stm32"
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/rogerclarkmelbourne/Arduino_STM32"
  }
}

Now `cd tools` and edit both `platformio-build-stm32f1.py` and `platformio-build-stm32f1.py`.
In both we need to change the line:
```python
FRAMEWORK_ROOT = platform.get_package_dir("framework-arduinoststm32-maple")
```
to
```python
FRAMEWORK_ROOT = platform.get_package_dir("framework-arduinoststm32-mapleUPDATED")
```

**DONE!** :nerd_face:
```

soreau, 2021-06-30:
Should work with latest.