arpruss/USBComposite_stm32f1 #57: MIDI and CompositeSerial not working together with latest release

3gyptian, 2020-03-06:
I'm having trouble running MIDI and CompositeSerial together with the latest release.  The same code works fine with older releases.  What happens on the newer releases is while the CompositeSerial communication works the midi is silent both on transmit and receive.  I'm finding Midi or Serial work ok configured independently with the later commits .. just not together.  

Rolling back through the git the last commit where serial and midi work together for me is here:

```
commit d5917495d0358fd457d1d7562f8f3f04489e7293
Author: arpruss <arpruss@gmail.com>
Date:   Fri Dec 20 21:14:56 2019 -0600

    foo
```

And I have no MIDI activity if I compile against any commits after that point .. before or after the merges:

[image]

Here's my test:

```
#include "USBComposite.h"
#include <Arduino.h>

uint8_t TX_CH=1;
uint8_t RX_CH=2;

uint8_t last_note=0;
uint8_t last_velocity=0;
uint8_t last_channel=0;

USBCompositeSerial mySerial;

class myMidi : public USBMIDI {
  virtual void handleNoteOff(unsigned int channel, unsigned int note, unsigned int velocity) {
    last_note=note;
    last_velocity=0;
    last_channel=channel;
  }
  virtual void handleNoteOn(unsigned int channel, unsigned int note, unsigned int velocity) {
    last_note=note;
    last_velocity=velocity;
    last_channel=channel;
  }
};

myMidi midi;

void pulseIt(int);

void setup() {
  
  pinMode(LED_BUILTIN,OUTPUT);

  USBComposite.clear(); // clear any plugins previously registered
  mySerial.registerComponent();
  midi.registerComponent();
  USBComposite.begin();
  USBComposite.setProductId(0x0031);  

  midi.begin();
  mySerial.begin(38400);

  while(!mySerial.isConnected()); // spin waiting for usb serial to come active
  delay(500);
  mySerial.print("TX Channel: ");mySerial.print(TX_CH); mySerial.print(" == MIDI Channel: ");mySerial.println(TX_CH+1);
  mySerial.print("RX Channel: ");mySerial.print(RX_CH); mySerial.print(" == MIDI Channel: ");mySerial.println(RX_CH+1);
  pulseIt(300);

} 

void loop() {

  //TX
  /*
  for (int i=0;i<5;i++) {
    midi.sendNoteOn(4, 68+i, 127);
    delay(50);
    midi.sendNoteOff(4, 68+i, 127);
    delay(50);
  }
  */

  // RX
  midi.poll(); // any midi coming in
  if (last_note!=0 && last_channel == RX_CH){ 
    mySerial.print(millis()); mySerial.print(" Note: ");mySerial.print(last_note);mySerial.print(" Velocity: ");mySerial.println(last_velocity); 
    last_note=0;
    pulseIt(200);
  }

}

void pulseIt(int pulseWidth ) {
  digitalWrite(LED_BUILTIN,LOW);
  delay(pulseWidth);
  digitalWrite(LED_BUILTIN,HIGH);
}
```

Perhaps this from something I'm not initializing correctly in my code?  

thanks

arpruss, 2020-03-07:
You should have USBComposite.setProductId(0x0031); before USBComposite.begin();.
But that shouldn't affect the issue in question.

arpruss, 2020-03-07:
I'm puzzled. The next commit after the one that you say works is 8c2d804b052f3cf2645ffdad177fa456fc176c34, and it doesn't change serial or midi.

3gyptian, 2020-03-07:
Well 8c2d804 is on another branch.  So working up from the beginning of that branch 
Serial and Midi are working at f7b28c5 but break at d279b20 where I find the serial port is not created on my mac and the midi device does appear but doesn't work.  The serial port returns on the commit following that (5636c41) but no midi response .. which is how the most recent commit behaves.  

[image]

fyi - as I mentioned on the other issue, I'm running maple core and modifying USB_EP_TYPE_ISO to USB_EP_T_ISO to get it to compile.  I'm not needing usb audio but perhaps this is causing an issue?   And I do have same issue with this definition and same results when testing with the STM32Duino *(default) core.

Here's my normal .platformio:

env:genericSTM32F103CB]
platform = ststm32
board = genericSTM32F103CB
framework = arduino
build_flags = -fno-threadsafe-statics
upload_protocol = stlink
;upload_protocol = dfu
;upload_port = 1eaf:0003
;upload_port = 1eaf:0031
;upload_port=/dev/u.usbmodem145201
monitor_speed = 115200
;build_flags = -fno-threadsafe-statics
;maple==roger clark
board_build.core = maple

thx

arpruss, 2020-03-08:
I don't really understand the git branch and commit system very well, I have to confess.
But if I understand you, you are saying that d279b20 broke the serial port, right? But d279b20 doesn't affect either midi or serial code, as far as I can tell, and only changes usbaudio, right?

3gyptian, 2020-03-09:
Ya I'm learning this git thing myself right now and it's tricky with branches as you have to make sure diffs are run against the previous commit on that branch and not just the previous commit which might be on a different branch.  I've been using 'git log --oneline --graph' to help see what's going on.  

So here's a snippet of what I believe is the area of commits where the midi/serial combo breaks .. with the most recent commit at the top:

```
...
Dec 21/19
*   a7c1f1d Merge branch 'master' into more_endpoints.  <== No midi after branches merged
|\  
| * d591749 foo       <== Master branch -- serial/midi ok 
| * 1295510 allow x360 endpoint number change    <== Master branch -- serial/midi ok
| * f7b28c5 parens around define       <<== Master branch -- serial/midi ok
* | 0440d69 explain endpoint maximum      <== Alternate branch -- midi still not working
* | 8c2d804 cleanup
* | 601cac2 bugfixes
* | 5636c41 avoid overlap in endpoint numbers between different parts
* | d279b20 bug fixes
* | 2601c99 update docs for more endpoints.         
* | d7887a7 make more endpoints available.     <== Alternate branch -- serial/midi not working
|/  
* 4330ab3 check interface number for x360 no data setup. <== Master branch -- serial/midi ok
* fdbcf10 set all xbox buttons at once
Dec18/19
...
```

I did a bunch more testing today after I upgraded the platformio maple core but other than the compile error being eliminated, as you can see above things are still broken on the non-master branch that was created on Dec18/19

The diff to do is between the first commit on the alternate branch d7887a7 and the previous commit before that on the master branch 4330ab3 .. and this diff shows a lot of updates against several file.  So  seemingly it's something starting here in this alternate branch that has broken the midi/serial on the master branch once these two are merge further along at a7c1f1d on Dec 21.

[image]

This is from 'Git History' plugin in Visual Studio code .. I find that interface works the best for comparing all these things.  

Happy to do more testing .. and thank you for your interest in resolving.

arpruss, 2020-03-09:
I either browse the commits on the GitHub website or use svn.

arpruss, 2020-05-13:
Register serial after registering midi and it should work.

It's a bug in the serial code. It also occurs if you composite serial and HID, with serial first.

arpruss, 2020-05-13:
Quick workaround: register midi first, then serial second.

arpruss, 2020-05-13:
Second workaround: 
```
#define MATCHING_ENDPOINT_RANGES
```
at the top of usb_generic.c

arpruss, 2020-05-13:
Fixed.

closed 2020-05-13 by arpruss

3gyptian, 2020-05-16:
rebuilt on 0.99 .. same code and works perfect!  Thanks a ton man.  Say..where do we send you beer $$?

arpruss, 2020-05-16:
Sorry for having taken so long to get to the fix. If you want to send something -- no obligation!! -- you can paypal to arpruss at gmail dot com.

arpruss, 2020-05-16:
Thank you so much for your too generous gift!

3gyptian, 2020-05-17:
you're welcome .. it's well earned!