arpruss/USBComposite_stm32f1 #62: How do you use the serial debugger after compositing a midi?

spherical-sound-society, 2020-05-08:
So when I run the midiout example the serial port get lost.

I can upload new sketches by resetting the blue pill but as long as the serial ort is missing I can´t use the serial,print commands to debug. How do you debug it?

Using stm32duino bootloader and USB, windows 10 64bit

That´s a great library. Thanks!!

arpruss, 2020-05-08:
I think this is a duplicate of https://github.com/arpruss/USBComposite_stm32f1/issues/57
Fixing this would require going through the commit history and seeing which commit broke this functionality.

3gyptian, 2020-05-09:
Yeah it would be nice to get this resolved somehow.  I’ve been living with building from this old commit as I posted on the other issue and all has been working well ..

https://github.com/arpruss/USBComposite_stm32f1/commit/f7b28c53ffebcdce2c63bcedc928e5d344338578

spherical-sound-society, 2020-05-09:
Arpruss, is the very midi out example that disables the serial.

Egiptian: ATM what I do is to have a isDebugging boolean at the beginning. If It´s debuging mode, I set up serial but not midi. If it´s not debugging, I set up Midi but not serial. This way at least I can debug the rest of the program, but I cannot debug the midi section.

Luckly the midi library works so nice that even without serial debugger you can use it 'cause is quite straightforward

If somehow we find a way for the usbcomposite not to shut down the serial would be awesome, but currently it kinds of works this way

Excuse my hurry language orthographs. 

Thanks!!

3gyptian, 2020-05-11:
hey .. yeah I was debugging like that too for a while but as mentioned earlier simply reverting to and older commit has serial and midi coexisting just fine. 

The following should get you to the same working version I'm using:

In your project's lib folder:
# git clone https://github.com/arpruss/USBComposite_stm32f1
# cd USBComposite_stm32f1
# git checkout f7b28c53ffebcdce2c63bcedc928e5d344338578

Note: if you're running PlatformIO and running Roger Clark's core (board_build.core = maple) then you'll likely run into a compile error which can be fixed by updating to Roger's latest core as per #55 .. then recompiling. 

lib/USBComposite_stm32f1/usb_audio.c:225:32: error: 'USB_EP_TYPE_ISO' undeclared here (not in a function); did you mean 'USB_EP_T_ISO'?

good luck!

spherical-sound-society, 2020-05-11:
Wow, didn´t catch that one. It looks fantastic... Downloading... Thank you s much Egiptian and Arpruss!

<3<3!!

closed 2020-05-11 by spherical-sound-society