arpruss/USBComposite_stm32f1 #63: Making Absolute Mouse HID to work on Android

ayavilevich, 2020-05-09:
Hey, I am emulating a composite device with a keyboard and AbsMouse using default descriptors and based on the example in
https://github.com/arpruss/USBComposite_stm32f1/blob/master/examples/absmouse/absmouse.ino

This works ok in Windows. The cursor moves and clicks happen thought there is an offset to x.
When I connect the device to an Android phone, the mouse doesn't seem to work while keyboard works ok.

Any idea if the default absolute mouse descriptor should work for Android?

I searched for some resources online and found different options for descriptors. Some mention touch screens, touch pads, digitizers, etc.

https://stackoverflow.com/questions/46033581/what-is-a-correct-usbhidreportdescriptor-for-android
https://stackoverflow.com/questions/11896471/usb-touchscreen-on-android-4-0-3-not-possible-to-select-only-moving-the-poin
https://stackoverflow.com/questions/13474017/android-usb-touchscreen-hid-descriptors-issue

There seem to be lots of different parameters and combinations. 
Can somebody help with a suggestion on how to continue? I am not familiar with the syntax of the descriptors. Thanks.

arpruss, 2020-05-10:
I don't have absolute mouse working on Android either.

ayavilevich, 2020-05-10:
Ok, thanks for your update.

I will then try to add a descriptor for a Digitizer device, as described in
https://www.codeproject.com/Articles/1001891/A-USB-HID-Keyboard-Mouse-Touchscreen-emulator-with

I am planning to define a new HID_[X]_REPORT_DESCRIPTOR macro and then create an instance of HIDReporter similar to HIDAbsMouse. Would that be enough or to I need to change something else?

arpruss, 2020-05-11:
I think so. For another simple example of a new HID device, see the simple two-axis two-button joystick in https://github.com/arpruss/paddlecontrollers