arpruss/USBComposite_stm32f1 #65: How to change MIDI device source name from Maple?

3gyptian, 2020-05-12:
My composite MIDI device is working perfectly however I'd like to change how it displays on the host as something different than Maple.  I've searched the library header files and elsewhere looking where this might be defined to no avail.  

Such a great library this is.  Thank you!

arpruss, 2020-05-13:
```
USBComposite.setVendorId(uint16 vendor);
USBComposite.setProductId(uint16 product);
USBComposite.setManufacturerString(const char* manufacturer);
USBComposite.setProductString(const char* product);
USBComposite.setSerialString(const char* serialNumber);
```

Windows may still show the old name if you just change the name without changing either the product or the vendor id.

closed 2020-05-13 by arpruss

3gyptian, 2020-05-16:
That totally worked for me.  Thanks man!!!