arpruss/USBComposite_stm32f1 #67: Keyboard cannot be used in BIOS

HelloDB, 2020-06-20:
Keyboard cannot be used in BIOS. It is normal in windows. What should I do

arpruss, 2020-06-20:
For BIOS use, you need to use BootKeyboard instead of Keyboard.

HelloDB, 2020-06-21:
> For BIOS use, you need to use BootKeyboard instead of Keyboard.

Thank you for your reply，I did use the bootkeyboard sample and the latest library files。
The SET_IDLE request of BIOS may not be responded correctly
```
#include <Arduino.h>
#include <USBComposite.h>

USBHID HID;
HIDKeyboard BootKeyboard(HID, 0);

void setup() 
{
    HID.begin(HID_BOOT_KEYBOARD);
    BootKeyboard.begin(); // needed just in case you need LED support
}

void loop() 
{
  BootKeyboard.press(KEY_RIGHT_ARROW);
  delay(100);
  BootKeyboard.release(KEY_RIGHT_ARROW);
  delay(1000);
  
  BootKeyboard.press(KEY_LEFT_ARROW);
  delay(100);
  BootKeyboard.release(KEY_LEFT_ARROW);
  delay(1000);
}

```
I tested the BIOS environment with the HID Library of Arduino Pro micro and NicoHood. It works normally, but it does not support STM32，https://github.com/NicoHood/HID

arpruss, 2020-06-21:
What is the SET_IDLE request? Alas I know nothing about the boot keyboard specs.

HelloDB, 2020-06-21:
> What is the SET_IDLE request? Alas I know nothing about the boot keyboard specs.

https://github.com/NicoHood/HID/blob/master/src/SingleReport/BootKeyboard.cpp
BIOS stage enumeration needs to respond to host SET_PROTOCOL and SET_IDLE and other commands

arpruss, 2020-06-21:
See if it works now. I added rudimentary SET/GET_IDLE support.

HelloDB, 2020-06-21:
Sorry, it still doesn't work

sparkdemcisin81, 2021-01-07:
any update on boot keyboard?

arpruss, 2021-01-07:
I have no idea what the problem is and hence how to fix it. Sorry.

sparkdemcisin81, 2021-01-07:
maybe you could check NicoHood lib to add it to your lib?

arpruss, 2021-01-07:
That library is very different.

I need someone to figure out exactly what is different about a boot keyboard from a normal keyboard, and then tell me. :-)

sparkdemcisin81, 2021-01-09:
here is some description of boot keyboard
https://www.usb.org/sites/default/files/hid1_11.pdf, page 74

sparkdemcisin81, 2021-01-09:
Boot Interface Descriptors at page 59