arpruss/USBComposite_stm32f1 #69: Need help for midi buttons

4dvn, 2020-07-16:
Hello,
I have 16 buttons midi notes from 36 to 51, can you give me the example of this?

4dvn, 2020-07-17:
> I am pretty sure the blue pill has enough GPIOs to read the buttons.

Is that possible to use button matrix 4x4

arpruss, 2020-07-17:
Yes, but it's not necessary.

4dvn, 2020-07-17:
> Yes, but it's not necessary.

I'm still stuck of this so i need you help because i just done it with arduino uno before that and i want to try with stm32

ecls-ecls, 2020-12-03:
If you are using arduino framework it is the same: 

set pin mode for each pin:

pinMode(PA5, INPUT_PULLUP); // Pulled up pin, triggered down when connected to GND, HIGH when left open. PA5 is just an example.
// do the same for every other pin.

And just read with digitalRead:
int pinState = digitalRead(PA5);
// do the same foreach other pin.

Remember when using a pulled up pin that when the button is pressed pin state will be LOW, inverted logic that you handle by software.

You can use a pulled down pin, and trigger with VCC (3.3 V or 5 V on pins that are 5V tolerant CHIP BURNING ALERT!, to be safe just use 3.3 V). In this case the logic won't be inverted when pin is HIGH the button is pressed.

If your pin is left open when the button is released, do not use a INPUT mode, as for open pins this may be a undetermined state, so pull down or up.

Bonus note:
When using mechanical triggers you may find that output signal is noisy, there are some ways on handlig this: you can use a schmidt trigger and a capacitor (more details on the web), or filter this by software. This is called debouncing.