arpruss/USBComposite_stm32f1 #75: boards.h: No such file or directory

mentaluproar, 2020-09-17:
When attempting to use this library on an SKR E3-DIP 3d printer control board, set as a genericSTM32F103RC board, I get a missing file error.  No boards.h file.  I figured I would have to configure this board somewhere but there's no boards.h file to modify.

arpruss, 2020-09-17:
I remember such a report. I think the person wasn't using the official Arduino IDE.

mentaluproar, 2020-09-18:
Oh sorry, I am using platformIO.  I wasn't able to get the Arduino IDE to do anything with this control board.

arpruss, 2020-09-18:
See: https://www.instructables.com/id/Gamecube-Controller-USB-Adapter-and-Getting-Starte/

BrianHarper, 2020-10-31:
I have a ST-link v2 and "bluepill), trying to program with Arduino 1.8.11 and the latest STM32 cores board config. But when trying to compile the MidiPotentiometer.ino I'm getting the following:-  
Arduino: 1.8.11 (Windows 7), TD: 1.51, Board: "Generic STM32F1 series, BluePill F103C8, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), HID (keyboard and mouse), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

In file included from C:\Users\user\Desktop\arduino-1.8.11\portable\sketchbook\libraries\USBComposite_stm32f1-master\examples\MidiPotentiometer\MidiPotentiometer.ino:1:
C:\Users\user\Desktop\arduino-1.8.11\portable\sketchbook\libraries\USBComposite_stm32f1-master/USBComposite.h:4:10: fatal error: boards.h: No such file or directory
    4 | #include <boards.h>
      |          ^~~~~~~~~~
compilation terminated.

exit status 1
Error compiling for board Generic STM32F1 series.

Could some one plese point me in the right direction as to what I am missing.
Thanks....

arpruss, 2020-10-31:
Are you using platformIO?

BrianHarper, 2020-10-31:
> 
> 
> Are you using platformIO?

No, using the Arduino 1.8.11 IDE

arpruss, 2020-10-31:
Looking at your error message, I think you're not using Roger's libmaple based core.

BrianHarper, 2020-10-31:
No I don't have that installed, just the standard "stm32 cores". do I need Roger's libmaple based core as well,  or instead of what I already have installed

arpruss, 2020-10-31:
This library is not compatible with the stm supplied core. You need the libmaple core instead.

closed 2020-10-31 by arpruss

BrianHarper, 2020-10-31:
Yes just installed a seperate portable IDE and installed the libmaple core and managed to get it working, Thanks for your help.
Now to build a USB Midi CC controller.....