arpruss/USBComposite_stm32f1 #77: PIO cannot compile the current library correctly,But the USBserial works correctly,

CanNev, 2020-10-10:
PIO cannot compile the current library correctly,But the USBserial works correctly,

PlatformIO Core 5.0.1·Home 3.3.0
------------------
System Information
------------------
      Time of this report: 10/10/2020, 08:27:04
             Machine name: DESKTOP-KRRHQQE
               Machine Id: {EE98EFE1-4895-492A-8B40-AA8A173EADB0}
         Operating System: Windows 10 专业版 64-bit (10.0, Build 18363) (18362.19h1_release.190318-1202)
                 Language: Chinese (Simplified) (Regional Setting: Chinese (Simplified))
      System Manufacturer: LENOVO
             System Model: 90FF0014CD
                     BIOS: LENOVO BIOS Rev:M0UKT23A 0.0 (type: BIOS)
                Processor: Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz (4 CPUs), ~3.2GHz
                   Memory: 8192MB RAM
      Available OS Memory: 8142MB RAM
                Page File: 7930MB used, 12500MB available
              Windows Dir: C:\Windows
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 96 DPI (100 percent)
       System DPI Setting: 96 DPI (100 percent)
          DWM DPI Scaling: Disabled
                 Miracast: Available, no HDCP
Microsoft Graphics Hybrid: Not Supported
 DirectX Database Version: Unknown
           DxDiag Version: 10.00.18362.0387 64bit Unicode

------------

n file included from .pio\libdeps\genericSTM32F103C8\USBComposite for STM32F1/usb_generic.h:11:0,
                 from .pio\libdeps\genericSTM32F103C8\USBComposite for STM32F1/USBComposite.h:6,
                 from src\main.cpp:14:
C:\Users\HH\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\usb\stm32f1/usb_reg_map.h: In function 'uint32* usb_pma_ptr(uint32)':
C:\Users\HH\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\usb\stm32f1/usb_reg_map.h:353:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     return (uint32*)(USB_PMA_BASE + 2 * offset);
Compiling .pio\build\genericSTM32F103C8\libf79\USBComposite for STM32F1\usb_midi_device.c.o
                                   ^
Compiling .pio\build\genericSTM32F103C8\libf79\USBComposite for STM32F1\usb_multi_serial.c.o
.pio\libdeps\genericSTM32F103C8\USBComposite for STM32F1\usb_audio.c:225:32: error: 'USB_EP_TYPE_ISO' undeclared here (not in a function); did you mean 'USB_EP_T_ISO'?
         .bmAttributes        = USB_EP_TYPE_ISO, /* isochronous, not shared */
                                ^~~~~~~~~~~~~~~
                                USB_EP_T_ISO
*** [.pio\build\genericSTM32F103C8\libf79\USBComposite for STM32F1\usb_audio.c.o] Error 1

CanNev, 2020-10-10:
the library only work with Arduino IDE.not work in PIO?

arpruss, 2020-10-10:
Yeah. Search through the closed issues for PlatformIO. I think it has the wrong version of the core or something like that.

closed 2020-10-10 by arpruss