arpruss/USBComposite_stm32f1 #80: Nor xbox360wireless nor multixbox360 works on Steamlink

ecls-ecls, 2020-12-03:
Tried both on a Steamlink (raspberry and android).

With multixbox360 I can se only one controller.
With xbox360wireless I get none controllers.

Thanks.

arpruss, 2020-12-03:
Does  it work with other programs, or with Windows?

closed 2020-12-03 by arpruss

ecls-ecls, 2020-12-03:
Yeah, on Windows it works very well.
On android I can't see the device, but on raspbian, linux itself may be seeing all four devices (/dev/input/js* tested with cat), but when I start steamlink, I see only one device, I'm no expert but maybe a OS itself limitation (one device per port)???
Very nice lib, by the way.
One more thing: is it possible to create a 3d mouse For CAD?

ecls-ecls, 2020-12-03:
PS.: Library updated to current version (2020-12-03).

arpruss, 2020-12-03:
Have you tried jstest on Linux?

I think I've used the library (I think the wireless version) on my RPI with an emulator or two.

As for 3D mice, the code behind this project can be easily adapted: https://www.instructables.com/SpaceBall-4000-Serial-to-USB-Adapter/

ecls-ecls, 2020-12-03:
It seems a problem with Steamlink, though. I have inspected the code, and the library reports vendor and product ID from Microsoft, maybe Steam tries some proprietary protocol that the library might not recognize. Is it possible to tweak it to report as some generic gamepad? Actually I don't need analog axes, I'm building an arcade joystick, and I have tried the twojoysticks example without success, also.

Thanks for the quick reply and 3d mouse example.

Regards!