arpruss/USBComposite_stm32f1 #84: Add an extra Z and Zrotate axis to joystick

SarwarN88, 2021-01-02:
Hi,
Thanks for such a great library
I am trying to add an extra 2 axis to the joystick library to the total of 8. I know your joystick library has only 6 axis + 1 hat. Is it possible to increase the axis to 8 by adding Z and Zrotate? For a few days I been trying to modify the USBHID.h to add an extra axis but the usb device get error massage when connecting to the PC. When I modify the Joystick.cpp to add Z axis it give me error and can't upload to my Maple mini. 
Here is modified section of the USBHID code:
`#define HID_JOYSTICK_REPORT_DESCRIPTOR(...) \
	0x05, 0x01,						/*  Usage Page (Generic Desktop) */ \
	0x09, 0x04,						/*  Usage (Joystick) */ \
	0xA1, 0x01,						/*  Collection (Application) */ \
    0x85, MACRO_GET_ARGUMENT_1_WITH_DEFAULT(HID_JOYSTICK_REPORT_ID, ## __VA_ARGS__),  /*    REPORT_ID */ \
	0x15, 0x00,						/* 	 Logical Minimum (0) */ \
	0x25, 0x01,						/*    Logical Maximum (1) */ \
	0x75, 0x01,						/*    Report Size (1) */ \
	0x95, 0x20,						/*    Report Count (32) */ \
	0x05, 0x09,						/*    Usage Page (Button) */ \
	0x19, 0x01,						/*    Usage Minimum (Button #1) */ \
	0x29, 0x20,						/*    Usage Maximum (Button #32) */ \
	0x81, 0x02,						/*    Input (variable,absolute) */ \
	0x15, 0x00,						/*    Logical Minimum (0) */ \
	0x25, 0x07,						/*    Logical Maximum (7) */ \
	0x35, 0x00,						/*    Physical Minimum (0) */ \
	0x46, 0x3B, 0x01,				/*    Physical Maximum (315) */ \
	0x75, 0x04,						/*    Report Size (4) */ \
	0x95, 0x01,						/*    Report Count (1) */ \
	0x65, 0x14,						/*    Unit (20) */ \
    0x05, 0x01,                     /*    Usage Page (Generic Desktop) */ \
	0x09, 0x39,						/*    Usage (Hat switch) */ \
	0x81, 0x42,						/*    Input (variable,absolute,null_state) */ \
    0x05, 0x01,                     /* Usage Page (Generic Desktop) */ \
	0x09, 0x01,						/* Usage (Pointer) */ \
    0xA1, 0x00,                     /* Collection () */ \
	0x15, 0x00,						/*    Logical Minimum (0) */ \
	0x26, 0xFF, 0x03,				/*    Logical Maximum (1023) */ \
	0x75, 0x0A,						/*    Report Size (10) */ \
	0x95, 0x05,						/*    Report Count (5) */ \
	0x09, 0x30,						/*    Usage (X) */ \
	0x09, 0x31,						/*    Usage (Y) */ \
        0x09, 0x32,                                              /*    Usage (Z) */ \
	0x09, 0x33,						/*    Usage (Rx) */ \
	0x09, 0x34,						/*    Usage (Ry) */ \
	0x81, 0x02,						/*    Input (variable,absolute) */ \
    0xC0,                           /*  End Collection */ \
	0x15, 0x00,						/*  Logical Minimum (0) */ \
	0x26, 0xFF, 0x03,				/*  Logical Maximum (1023) */ \
	0x75, 0x0A,						/*  Report Size (10) */ \
	0x95, 0x02,						/*  Report Count (2) */ \
	0x09, 0x36,						/*  Usage (Slider) */ \
	0x09, 0x36,						/*  Usage (Slider) */ \
	0x81, 0x02,						/*  Input (variable,absolute) */ \
    MACRO_ARGUMENT_2_TO_END(__VA_ARGS__)  \
    0xC0`

`// only works for little-endian machines, but makes the code so much more
// readable
typedef struct {
    uint8_t reportID;
    uint32_t buttons;
    unsigned hat:4;
    unsigned x:10;
    unsigned y:10;
    unsigned z:10;
    unsigned rx:10;
    unsigned ry:10;
    unsigned sliderLeft:10;
    unsigned sliderRight:10;
} __packed JoystickReport_t;

static_assert(sizeof(JoystickReport_t)==15, "Wrong endianness/packing!");` // if I use 13 it give me error so I used 15. 

` void setManualReportMode(bool manualReport); // in manual report mode, report only sent when send() is called
    bool getManualReportMode();
	void begin(void);
	void end(void);
	void button(uint8_t button, bool val);
	void X(uint16_t val);
	void Y(uint16_t val);
        void Z(uint16_t val);
	void position(uint16_t x, uint16_t y);
	void Xrotate(uint16_t val);
	void Yrotate(uint16_t val);
	void sliderLeft(uint16_t val);
	void sliderRight(uint16_t val);
	void slider(uint16_t val);
	void hat(int16_t dir);
	HIDJoystick(USBHID& HID, uint8_t reportID=HID_JOYSTICK_REPORT_ID) 
            : HIDReporter(HID, hidReportJoystick, (uint8_t*)&joyReport, sizeof(joyReport), reportID) {
        joyReport.buttons = 0;
        joyReport.hat = 15;
        joyReport.x = 512;
        joyReport.y = 512;
        joyReport.z = 512;
        joyReport.rx = 512;
        joyReport.ry = 512;
        joyReport.sliderLeft = 0;
        joyReport.sliderRight = 0;
    }`

And I have added this section to the Joystick.cpp but it give me error and can't upload to the Maple mini:
`void HIDJoystick::Z(uint16_t val){
  if (val > 1023) val = 1023;
    joyReport.Z = val;
      
    safeSendReport();
}` 
I appreciate your help.

arpruss, 2021-01-02:
What error did you get?

SarwarN88, 2021-01-03:
As I said the usb joystick get recognised but in the windows 10 control panel it shows Maple with yellow exclamation mark and when I move the mouse cursor over it it shows  status: device error. In the game controller settings window the device is not even showing. It is only happen when I try to add another extra axis. I have tried the examples from your library like (simpleJoystick and signedJoystick) but they both have similar problem. I really need those two extra axis otherwise I will end up using 3 Maple minis for my flight simulator throttle controller. Your library is the only one that works perfectly for my device. It just need those 2 extra axis. I don't know why you didn't included Z and Zrotation axis in your library since windows game controller can see maximum of 8 axis +1 hat? I thought its a problem imbedded in the USBHID.h file? 
Thanks again

arpruss, 2021-01-03:
If you've added 10 bits of info, the report length is no longer a multiple of 8 bits. To fix that, you need to add 6 bits of padding at the end of the report. Put these before your End Collection and see if it helps:
  0x75, 0x06,                    /*   REPORT_SIZE (6) */ \
  0x95, 0x01,                    /*   REPORT_COUNT (1) */ \
  0x81, 0x03,                    /*   OUTPUT (Cnst,Var,Abs) */ \

SarwarN88, 2021-01-04:
OMG it worked! Thanks man you are my hero. I can't believe after all this many days of pulling my hair it was all about those remining bits that was not included. I readied somewhere in the original USBHID report descriptor that the data need to be sent in 8bits or 1 byte for each axis and the trick about the multiplayer method but I didn't understand it until you told me;)
[image]

SarwarN88, 2021-01-04:
Just a final question: I have changed the device name using vendor ID and product ID so my game can see the new device name rather than just Maple using the fuctions below. However, I can't put letter in vendorid or productid section or I will get error. I have to use for example 0x203 to get M2 name or wired sign name that I showed below. How I can get the exact name that I want for example (Rudder Pedals)? I tried using USB ID Database from online for a few Rudder Pedals but the names are all wired signs and short. Thanks again   
 const char ManufacturerName[] = "Sarwar";
const char DeviceName[] = "Rudder Pedals";
const char DeviceSerial[] = "00000000000000000001";
USBComposite.setManufacturerString(ManufacturerName);
USBComposite.setProductString(DeviceName);
USBComposite.setSerialString(DeviceSerial);
USBComposite.setVendorId(0x203);
USBComposite.setProductId(0x203);

[image]

arpruss, 2021-01-12:
The vendor ID and device ID are four hexadecimal digits each.

SarwarN88, 2021-01-12:
I am trying to change the name in the controller properties to lets say (Rudder Pedals).
In hex format this will be: 52 75 64 64 65 72 20 70 65 64 61 6c 73 0d 0a. How I can do this?

arpruss, 2021-01-12:
This should work, but it must be called before the begin call:
USBComposite.setProductString("Rudder Pedals");

Allexin, 2023-04-23:
@SarwarN88 i am trying to achive same goal: add ZRotation to report, but can't: all looks fine, but reports not working.
Can you provide some valid code? Full Descruptor and report? Thank you