arpruss/USBComposite_stm32f1 #90: Differentiate multiple Joysticks

LucasFeliciano21, 2021-03-07:
Hi guys, 

I'm already using the library and It's amazing, but I have a little issue that I hope you guys could help me with. 

I'm building a Hotas and for that, I'm using a total of 40 buttons and 10 axes. If DirectX could read that on windows I could just call off the day, but even changing the descriptors and been able to use the data on a nodejs web-based application, windows can't see my axes ( the ones above axis 6). So I returned to my original idea.

I've used 2 HID joysticks, so I could have up to 6 axes on each one. The problem then is to differentiate both devices. most applications differentiate by name, vendorID, or productID, which makes it very complicated to set different sets of buttons as they will be saved by the game as belonging to the same joystick(because of the name and Pid).

It's possible in a way to differentiate both devices with those conditions?

Many thanks.

arpruss, 2021-03-07:
I don't see how to do it. Maybe you could vary the number of buttons and tell them apart that way?

LucasFeliciano21, 2021-03-07:
The problem is that the game does the distinction by devId or name. 

as both devices are linked by the same name and devId, it always considers that was the same controller and I need to set up it all over again. 

I don't see any alternative. I'll try to see if I can write a script for that, but then, it's so annoying that windows don't accept the HID for joystick fully.