arpruss/USBComposite_stm32f1 #96: Plase make port for official STM Cores

baniaque, 2021-05-16:
PLEASE! :) Most of libs are focused to official cores and this make impossible to use your library in many cool projects.

madias123, 2022-03-08:
Since no "real" STM32F103Cx isn't available anymore, I have to change completely  to the F401/F411 pills, this this superb library ist getting obsolete.

arpruss, 2022-03-08:
The library works fine with the fakes that I've used.

arpruss, 2022-03-08:
I don't have the time or the need to figure out how the official core works. I would love it if someone ported this library.

twisted-electrons, 2023-06-25:
Hello, has there been any progress porting this to the official core? I'd love to know if there is a solution to get USB MIDI working on the STM32 official core? Thanks!

arpruss, 2023-06-25:
Nobody seems interested in picking up the task. And I don't use the official core as I'm told it's bloated.

twisted-electrons, 2023-06-25:
@arpruss Hi Thanks for your reply.

BennehBoy, 2023-09-28:
For anyone revisiting this topic...

I had a stab at porting this but it proved to be a bit of a Sisyphean task  - there's too much of the USB stack embedded in the official core which would require multiple PR's to interface with it... most of that is based on the bog standard HAL implementation which is really rather thin in content.  Another considerable issue in that space is that there are multiple other PRs for USB HID that conflict with one another, I just don't think that @fpistm has the time to refactor any of the USB side of things. So another PR would likely similarly languish in the repo.

fpistm, 2023-09-28:
@BennehBoy 
> I just don't think that @fpistm has the time to refactor any of the USB side of things. So another PR would likely similarly languish in the repo.

I would like refactor USB to be inline with Arduino implementation (Pluggable USB,...) and probably use TinyUSB. But currently time is always the missing ingredient.

arpruss, 2023-09-28:
I haven't looked at the official core, but if it exposes enough mid-level APIs, it shouldn't be super hard. On my library's side, I  once put a lot of time into refactoring it so that all the core-dependent code is in usb_generic.c. So at most one just has to rewrite those 918 lines of code. 

I suggest setting up a bounty, and then hopefully someone will be found to do it. Maybe even me. :-)

twisted-electrons, 2024-06-25:
@arpruss

Hi! Hope all si well

I have two STM32F4 projects in PlatformIO that desperately need USB MIDI support. I need this to work on the STM32F405. You mentioned setting up a bounty—I'm happy to do that, but could you please explain how?

Thank you!