arpruss/USBComposite_stm32f1 #98: USBXBox360 not working with XBox 360

Allexin, 2021-08-04:
I am use USBXBox360 to emulate XInput gamepad. 
It works fine on PC(Windows)
But when i try to use it with XBox 360 console - i am failed: console ignore my input.
Is it normal? How can i fix it?

arpruss, 2021-08-04:
Well, it's not a real XBox controller. For instance, it doesn't support the built-in mic. So I am not surprised that a console is more picky.

Allexin, 2021-08-04:
Maybe you can route me to the right direction:
i want to create USB Proxy device for gamepad.
I connect my device between console and real gamepad and just want to add little modfication to input.
please, if you can, provide some information how i can do it.

arpruss, 2021-08-05:
You need a microcontroller that does both USB host and USB device. This one only does USB device.

Allexin, 2021-08-05:
I am use MAX3421E usb host controller connected to STM32.
I think its offtop, so i am close issue.
Thank you for you work.

Allexin, 2021-08-06:
Its not working because XBox360 check decive compatibility through crypto requests.
Its name "Xbox Security Method 3"
I am newbie in so low coding. I found protocol specification. I want to resend cyprto requests from console to original gamepad.
But for this i must know when this requests is received by stm32. How i can intercept this requests from USBComposite?

closed 2021-08-09 by Allexin