arpruss/USBComposite_stm32f1 #106: Need help to modify Joystick

meilleur102, 2022-03-15:
Hi! 
I really struggle to understand the HUI portion of this board. 
I want to add more slider to the joystick library. 

I want to have multiple rotary encoder so it's the only way (I think) I can do this with this library. 
Can somebody help me with this. 
Thanks.

arpruss, 2022-03-16:
You need to modify the HID descriptor (the current one is in USBHID.h) for the joystick to include another axis, add the new axis to the JoystickReport_t structure, and add the interface for it to the Joystick class.

meilleur102, 2022-03-16:
Hi. Thx. 
I figure it out with button for the rotary encoder. I think ill be ok with just 2 slider. 
Quick question. Is there a place where I can find the pinNumber instead of using the pinName. 
When i built a struct I need to go with byte and not char so PB9 dont work. 
With the pinout of the bluepill can I be sure that they all match? 

Thx

arpruss, 2022-03-16:
I don't see why you can't put PB9 in a byte. Assuming it's only a byte long, which I am not 100% sure of.

meilleur102, 2022-03-16:
Can't tell you why. I'm a c# programmer so my experience on Arduino / microcontroller is very limited. 
For this project the HID is really new to me and I found a code for a Arduino Micro. 
I modify a couple of thing but the struct need byte and if I put PB9 the code give me a error when I try to do a digitalRead over the struct. 

Here a part of the code maybee you will understand more why. 

```
struct rotariesdef {
  byte pin1;
  byte pin2;
  int ccwchar;
  int cwchar;
  volatile unsigned char state;
  int btn;
};

rotariesdef rotaries[NUMROTARIES] {
  {9,10,24,25,0,1},
  {15,19,26,27,0,3},
  {20,21,28,29,0,5},
  {22,23,30,31,0,7},
};
```

If I put PB9 instead of 9 I have a error in this part of the code.

```
unsigned char rotary_process(int _i) {
   unsigned char pinstate = (digitalRead(rotaries[_i].pin2) << 1) | digitalRead(rotaries[_i].pin1);
  rotaries[_i].state = ttable[rotaries[_i].state & 0xf][pinstate];
  return (rotaries[_i].state & 0x30);
}
```

To be fair I dont even understand 80% of this code. 
There a lot of define with some hex and i'm lost. 

Anyways, the number seems to be ok for the moment. But if theres a way to put PB9 instead of 9 it will be nice. 

Last thing. I have maple bootloader on the chip but in Arduino I select Generic STM. When I go on device manager I don't see the USB device. But when I go in Game Controller I can see it and the button asigment work. 
The Game controller name is Maple. Can I change that? 
Thx.

arpruss, 2022-03-17:
1. What's the error message you get? You can always explicitly typecast to byte: (byte)PB9.
2. Call USBComposite.setManufacturerString(name) before starting up the USB device to set a different name. Note, however, that Windows will probably remember the previous name. You might also want to do setProductId() and setVendorId().

meilleur102, 2022-03-22:
> 1. What's the error message you get? You can always explicitly typecast to byte: (byte)PB9.
> 2. Call USBComposite.setManufacturerString(name) before starting up the USB device to set a different [...]

If I go with the "arduino" pin number it's working. So Ill go with that. 
Do you think it will be possible to use the Keymap library on the STM. 
Doing a Matrix is really easy with this library. 
Or if you can fwd me to some kind of tutorial for a matrix with your library that will be awesome. 
Thx

closed 2022-03-31 by meilleur102