arpruss/USBComposite_stm32f1 #109: Help with flashing

sonik-br, 2022-03-30:
Hi.

I flashed the `keyboardwithleds` sketch as it does have a serial port. Flashed via serial usb adapter.
Now I'm trying to flash via usb using upload method `STM32duino bootloader`
It does not find the board. Even if i keep it disconnected and only connected when I see the message `Searching for DFU device`.

Board is a bluepill. Both jumpers are now in position zero.

Need to change anything on the sketch to make the bootloader work?

sonik-br, 2022-03-31:
Com port is seleted on arduino ide. I can even open the serial monitor and get what the device is sending.
Both jumpers must be in position zero?

arpruss, 2022-03-31:
You can also just hit the reset button just before the upload starts.

sonik-br, 2022-03-31:
Don't work.

maple_loader v0.1
Resetting to bootloader via DTR pulse
Reset via USB Serial Failed! Did you select the right serial port?
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

Searching for DFU device [1EAF:0003]...
dfu-util - (C) 2007-2008 by OpenMoko Inc.
This program is Free Software and has ABSOLUTELY NO WARRANTY

Couldn't find the DFU device: [1EAF:0003]

sonik-br, 2022-03-31:
I'm new to this. Looks like I need to flash the custom bootloader right?
On Roger's repo says that I need to "fix" the bluepill to support the bootloader.
Looks like the HID Bootloader works on the bluepill without any mod.
Do you have any experience with it?

By the way, thanks for this amazing library! I've got dinput/xinput/switch output selectable by holding a button during boot. :)

closed 2022-03-31 by sonik-br