arpruss/USBComposite_stm32f1 #114: boards.h not found regardsless of platform

Dantali0n, 2022-07-04:
I have tried using this library both through the Arduino IDE and platformio and always get the same error:
`boards.h no such file or directory`.

System:
Arch Linux on btrfs fileystem (case sensitive fs)
Arduino IDE 1,8.19
Library 0.63.0

```
fatal error: boards.h: No such file or directory
 #include <boards.h>
```

Dantali0n, 2022-07-04:
I have the Arduino IDE working now.

Maybe we can try to get PlatformIO fixed?

One of the things it seems to dislike is the none semantic versioning versions of the library (1.06 instead of 1.0.6).

```
[env:bluepill_f103c8]
platform = ststm32
; https://github.com/stm32duino/Arduino_Core_STM32/blob/4be5ca1d770d2fab99a97914471fd548e9583609/tools/platformio/platformio-build.py
build_flags = -DPIO_FRAMEWORK_ARDUINO_ENABLE_HID

lib_deps = arpruss/USBComposite for STM32F1

board = bluepill_f103c8
; change microcontroller
board_build.mcu = stm32f103c8t6
; change MCU frequency
board_build.f_cpu = 72000000L
```

Dantali0n, 2022-07-04:
It did resolve a parsing error that was preventing me from selecting a specific version through the platformio.ini but it does not resolve `boards.h no such file`.

I have found that simply removing these imports gets us to a next stage (I think the inclusion of boards.h is managed automatically as other libraries do not do this either, some example: https://github.com/MHeironimus/ArduinoJoystickLibrary) instead they seem to include Arduino.h.

When I change these imports I get an error that it can' find libmaple but this is likely an issue for https://github.com/stm32duino/Arduino_Core_STM32 instead.

Dantali0n, 2022-07-04:
Seems for stm32duino you simply need 'board_build.core = maple`

Dantali0n, 2022-07-04:
I have a working configuration now, for some reason the order seems to matter as it works even with `include boards.h` now. If I change the order of arguments in platformio.ini it sometimes breaks. I can't figure this out:

```
[env:bluepill_f103c8]
platform = ststm32
; https://github.com/stm32duino/Arduino_Core_STM32/blob/4be5ca1d770d2fab99a97914471fd548e9583609/tools/platformio/platformio-build.py
; build_flags = -DPIO_FRAMEWORK_ARDUINO_ENABLE_HID -DPIO_FRAMEWORK_ARDUINO_STANDARD_LIB

lib_deps = arpruss/USBComposite for STM32F1 @ ^1.0.7

board = genericSTM32F103C8 ;bluepill_f103c8
; change microcontroller
board_build.mcu = stm32f103c8t6
; change MCU frequency
board_build.f_cpu = 72000000L
; we need
board_build.core = maple

framework = arduino

upload_protocol = serial
```

Dantali0n, 2022-07-05:
I think we can close this now, it seems to work for both `board = genericSTM32F103C8` and `board=bluepill_f103c8` at least. Maybe we could hint at:

 ```
board_build.core = maple
lib_deps = arpruss/USBComposite for STM32F1 @ ^1.0.7
```
In the readme that be nice I think

closed 2022-07-05 by Dantali0n

tinyboxxx, 2024-09-19:
thanks a lot

> I think we can close this now, it seems to work for both `board = genericSTM32F103C8` and `board=bluepill_f103c8` at least. Maybe we could hint at:
> 
> ```
> board_build.core = maple
> lib_deps = arpruss/USBComposite for STM32F1 @ ^1.0.7
> ```
> 
> In the readme that be nice I think

> I think we can close this now, it seems to work for both `board = genericSTM32F103C8` and `board=bluepill_f103c8` at least. Maybe we could hint at:
> 
> ```
> board_build.core = maple
> lib_deps = arpruss/USBComposite for STM32F1 @ ^1.0.7
> ```
> 
> In the readme that be nice I think

thanks a lot