arpruss/USBComposite_stm32f1 #116: Various libmaple headers not found when compiling against latest STM_32 boards release

bakoontz2, 2022-07-23:
Errors all referring to STM32F1 headers not being found:

```
In file included from C:\Users\588673\Documents\Arduino\libraries\USBComposite/USBComposite.h:6,
                 from C:\Users\588673\Documents\AmateurRadio\VNA\VNA\VNA\flashstruct.cpp:28:
C:\Users\588673\Documents\Arduino\libraries\USBComposite/usb_generic.h:3:10: fatal error: libmaple/ibmaple_types.h: No such file or directory
    3 | #include <libmaple/ibmaple_types.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
```
I am running with version 0.91.

arpruss, 2022-07-24:
I don't think you're using Roger's libmaple based core, but the official stm core (which is less efficient and not compatible with this library).

bakoontz2, 2022-07-25:
So somehow I ended up installing the "official" STM core, which apparently overrides the libmaple core in typical random Arduino IDE fashion.  Deleting this "update" seems to have fixed the issue, sorry for the false alarm!

closed 2022-07-25 by bakoontz2