arpruss/USBComposite_stm32f1 #117: Struggling to composite xinput with mouse

leecyrille, 2022-07-27:
I can't seem to make this work. As soon as I call something for the mouse it quits working. Do you have a recommendation?

`#include <USBComposite.h>

USBHID HID;

//USBMultiXBox360<4> x360;

void setup(){
  USBComposite.clear();
  Mouse.registerComponent();
  //x360.registerComponent();
  USBComposite.begin();
  USBComposite.setProductId(0x0031); 
  Mouse.begin();
  while (!USBComposite);
}

void loop(){
  static uint32_t previousMillis;
  uint32_t currentMillis = millis();
  const uint32_t interval = 1000;
  static int state; //used to indicate state 0 or state 1
  // States
  // 0 = wait for next state
  // 1 = do action 1
  // 2 = wait for next state
  // 3 = do action 2 & set state to 0

  
if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    state++;
  }

if (state == 1)
  {
    //x360.controllers[0].X(32767);
    //x360.controllers[1].Y(32767);
    //x360.controllers[2].XRight(32767);
    //x360.controllers[3].YRight(32767);
    Mouse.move(20,0);
    state = 2;
  }

if (state == 3)
  {
    //x360.controllers[0].X(-32768);
    //x360.controllers[1].Y(-32768);
    //x360.controllers[2].XRight(-32768);
    //x360.controllers[3].YRight(-32768);
    Mouse.move(-20,0);
    state = 0;
  }
  
}#include <USBComposite.h>

USBHID HID;

//USBMultiXBox360<4> x360;

void setup(){
  USBComposite.clear();
  Mouse.registerComponent();
  //x360.registerComponent();
  USBComposite.begin();
  USBComposite.setProductId(0x0031); 
  Mouse.begin();
  while (!USBComposite);
}

void loop(){
  static uint32_t previousMillis;
  uint32_t currentMillis = millis();
  const uint32_t interval = 1000;
  static int state; //used to indicate state 0 or state 1
  // States
  // 0 = wait for next state
  // 1 = do action 1
  // 2 = wait for next state
  // 3 = do action 2 & set state to 0

  
if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    state++;
  }

if (state == 1)
  {
    //x360.controllers[0].X(32767);
    //x360.controllers[1].Y(32767);
    //x360.controllers[2].XRight(32767);
    //x360.controllers[3].YRight(32767);
    Mouse.move(20,0);
    state = 2;
  }

if (state == 3)
  {
    //x360.controllers[0].X(-32768);
    //x360.controllers[1].Y(-32768);
    //x360.controllers[2].XRight(-32768);
    //x360.controllers[3].YRight(-32768);
    Mouse.move(-20,0);
    state = 0;
  }
  
}l`

arpruss, 2022-07-27:
Have you looked at the demo mouse code?

leecyrille, 2022-07-27:
> Have you looked at the demo mouse code?

I tried just moving those parts over, but I guess that's a basic example. it didn't work.