arpruss/USBComposite_stm32f1 #118: Incompatibility with other libraries

SunboX, 2022-09-19:
First, thanks for this great library! :)

Many libraries nowadays are using the STM32Duino core instead of the Maple core. Regarding this thread [1], the Maple core seems also to be outdated and difficult to set up.

Is there any plan to move the Library to STM32Duino core?

[1] https://community.platformio.org/t/stm32-platform-uploads-elf-file-not-bin/29065/14

SunboX, 2022-09-19:
ah, just found this issue: https://github.com/stm32duino/Arduino_Core_STM32/issues/668

alex-eri, 2023-10-21:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
upload_flags = -c set CPUTAPID 0x2ba01477
debug_server = /home/eri/.platformio/packages/tool-openocd/bin/openocd
        -s /home/user/.platformio/packages/tool-openocd/scripts
        -f interface/stlink.cfg
        -c "transport select hla_swd"
        -c "set CPUTAPID 0x2ba01477"
        -f target/stm32f1x.cfg
        -c "reset_config none"
lib_deps =
        SPI
        arpruss/USBComposite for STM32F1@^1.0.9
debug_init_break =
board_build.ldscript = jtag_c8.ld
```

My setup for f103c8 clone in platformio. So to setup start with:

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
```