arpruss/USBComposite_stm32f1 #121: Joystick with bluepill clone not showing up on Joy.cpl in other devices

Mechics, 2022-10-11:
Hi,
I am using an STM32F103C8T6(fake) Blue pill. It has 1.5k at r10. But it still doesn't connect to flash loader demonstrator. As i have a working stlink v2, i overcame this by connecting it to st link utility and flashing the bootloader binary.

As i want to use the MC as a joystick. I used Arpruss composite library and followed steps from this instructable: https://www.instructables.com/Gamecube-Controller-USB-Adapter-and-Getting-Starte/

But i wasnt able the code to upload this via DFU. so i gave up on that and moved to platform.io
In platform.io i realised that  the MC is a FAKE with id 0x2b01477.
So i adjusted the platform.ini folder  to this.

```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
board_build.core = maple
upload_flags = -c set CPUTAPID 0x2ba01477
lib_deps =
    arpruss/USBComposite for STM32F1@^1.0.7
 ```   
 

 I used the same code i the instructable and uploaded the code to blue pill via STLINK and it worked.

```

#include <USBComposite.h>
#include <libmaple/usb.h>

USBHID HID;
HIDKeyboard Keyboard(HID);
HIDJoystick Joystick(HID);
HIDMouse Mouse(HID);

void setup() {
  USBHID.begin(HID_KEYBOARD_MOUSE_JOYSTICK);
  while (!usb_is_connected(USBLIB) || !usb_is_configured(USBLIB)) delay(100);
  Keyboard.println("Hello world!");
}

void loop() {
  Joystick.X(0);
  Joystick.Y(0);
  digitalWrite(PB12, 0);
  delay(500);
  Joystick.X(1023);
  Joystick.Y(1023);
  delay(500);
  digitalWrite(PB12, 1);
}

```

The bluepill showed up as “maple” in my Joy.cpl when i connected it with the computer i uploaed the code from using a USB cable.
But the issue is that when i connect my bluepill to any other computer it doesnt show up in joy.cpl instead it  shows up as Unknown usb device (Device descripter request failed).

 However it works perfectly on the system i am using. i have tried installing drivers mentioned in the instructable on other computers but that also doesn't work.

Do anyone know why this is happening and any possible solutions?
Thank you.

arpruss, 2022-10-11:
Try changing the usb device ID.

Mechics, 2022-10-13:
Hi,
i tried doing this
```
void setup() {
//USBComposite.setManufacturerString(ManufacturerName);//controller rename
//USBComposite.setProductString(DeviceName);//controller rename
//USBComposite.setSerialString(DeviceSerial);//controller rename
USBComposite.setVendorId(0x987);//controller rename
USBComposite.setProductId(0x46);
  HID.begin(HID_KEYBOARD_MOUSE_JOYSTICK);
  while (!usb_is_connected(USBLIB) || !usb_is_configured(USBLIB)) delay(100);
}
```
in the main program. Is this what you meant. This didnt do anything as the USB is still not getting recognised.

Allexin, 2023-04-23:
Did you check R10 on board? I spend a lot of time, and wreck some boards before i am realize: the problem is in the hardware. Add right resistor to board and now all works like a charm.