arpruss/USBComposite_stm32f1 #131: How to use the midi.sendPitchChange function?

risyanto88, 2024-01-15:
In midicomposit there is a function midi.sendPitchChange(). What is a simple example to carry out this function. I have tried entering the 16 bit code from 0 to 127 manually, but the results are not as desired. Is it possible that the midi.sendPitchChange() library doesn't work? Or is there still a bug in this function? So a simple example is not included to explain.

risyanto88, 2024-01-15:
I am using stm32f103c8t6 for my midi project, using this library as the midi pitch control, are there any other instructions that can be used for midi pitchBend that can be used with this library? Please give me instructions so that my project can be perfect

risyanto88, 2024-01-15:
> I have no idea. I suggest looking up MIDI specs and looking at the code.

Are you also using the stm32f103c8t6 board for your midi projects?

arpruss, 2024-01-15:
Yes, but my one project was very simple: just play notes when touches on the device ports are detected.

risyanto88, 2024-01-15:
Didn't you create this library? I'm still curious about the midi.sendPitchChange function in the USBMIDI.cpp file It explains that you have to use 14 bit code, can this be used correctly? Can you give some suitable examples?

// Send a Midi PITCH CHANGE message, with a 14-bit pitch (always for all channels)
void USBMIDI::sendPitchChange(unsigned int pitch)
{
    outPacket.p.cable=DEFAULT_MIDI_CABLE;
    outPacket.p.cin=CIN_PITCH_WHEEL;
    outPacket.p.midi0=MIDIv1_PITCH_WHEEL;
    outPacket.p.midi1= (uint8) pitch & 0x07F;
    outPacket.p.midi2= (uint8)  (pitch>>7) & 0x7f;
    writePacket(outPacket.i);
    
}

arpruss, 2024-01-15:
I reused standalone MIDI code from someone else. The LICENSE file says:
MIDI code Copyright (c) 2010 Perry Hung, (c) 2013 Magnus Lundin, (c) 2013 Donald Delmar Davis, Suspect Devices,
(c) 2003-2008 Tymm Twillman <tymm@booyaka.com>.

tymmothy, 2024-01-16:
It's been a long time since I've touched this code and there were definitely some bugs when I last did anything with it, but in general I think the current code here is generally correct - however note that the MIDI standard and common usage of the parameter may vary.

note: i did some quick dumps with midisnoop that appear to match my expectations here before responding.

Specifically, see note 4 here: https://sites.uci.edu/camp2014/2014/04/30/managing-midi-pitchbend-messages/ -- value 8192 maps to no bend; smaller values than 8192 map to bend down, larger to bend up. Many receivers will convert to a signed value and subtract 8192 from the value before using it, showing a range -8192 to 8191.

(the value in the packet is 14 bits, which means the complete unsigned range is from 0-16383).

so if you're passing in values from 0 to 127, you would end up in many cases go from seeing a very large downward bend, to an only slightly less large downward bend.

risyanto88, 2024-04-15:
I have successfully run the sendPitchChange function, but only to move right and left or the x axis, how does the y axis function? Are there any special commands in the library that I need to run?

closed 2024-09-23 by risyanto88