arpruss/USBComposite_stm32f1 #134: boards.h not found

Michri, 2025-01-15:
When compiling in Arduino IDE, I get an error:

USBComposite.h:4:10: fatal error: boards.h: No such file or directory
    4 | #include <boards.h>

Where may I find this?

adampisula, 2025-01-31:
The same error occurs on PlatformIO `v6.1.16`.

arpruss, 2025-02-01:
You aren't by any chance using the official stm32 board support? You need to use Roger's libmaple one.

adampisula, 2025-02-01:
Yeah I noticed that yesterday too. Thanks for clarifying anyway!

For future reference: there are multiple STM32 Arduino cores, and this library handles USB for Roger Melbourne's Maple core (https://github.com/rogerclarkmelbourne/Arduino_STM32). I was using STM32duino, which is a different, STM32-official core. There seems to be a USB library for that implementation but I have yet to test it.

Cheers!